Klipper3d/klipper #2846: DRV8825 stepper drivers

Hairyloon, 2020-05-08:
Are these compatible with Klipper?
Is there something needs setting in the config to make them work properly? Apart from the step distance.
The motors are getting very hot and not running properly, but they are stepping about the right distance.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4598609/klippy.log)

theopensourcerer, 2020-05-08:
They work fine for me. I still have one left in any case  - on my extruder.

By the sounds of it you need to adjust the pot on the driver to get the current right for your stepper. I don't think it's anything to do with Klipper per-se.

Hairyloon, 2020-05-08:
Yes, I have tried that, but I'm working a bit blind as I don't know what I'm aiming for.
I've turned them down 'till I met an endstop on the trimmer (or that's what it felt like) and it made an improvement, but not sorted it.

manu7irl, 2020-05-08:
You have to know what kind of stepper you have and aim for the amperage per
phase on each of them.
Drv8825 have the vref to be calibrated like this:
Motor_amp/0.5=vref
I myself to be safe remove 10% of the value, I got from this.

Hairyloon, 2020-05-08:
I'd read that, but measuring where I thought I was supposed to I was getting 10-12V... Which may suggest the main board is faulty.
Or I'm measuring in the wrong place... that seems more likely.

manu7irl, 2020-05-08:
https://youtu.be/YJSmrbeZUp4
Start with this but do it carefully, or you might burn some drivers.

Hairyloon, 2020-05-08:
Thanks, I'll give that a go, but judging by the heat of the motors when they're sitting doing nothing, I think if the drivers were going to burn, they'd have burned by now, but I do have one spare.
I'm not entirely convinced it's the pots though: running Marlin doesn't seem to have the same effect.  Ive not run it for long though as it wasn't working & I was called away: it may simply be that it didn't have time to heat up.
I will try again in due course.

Hairyloon, 2020-05-08:
Ok, progress: it is printing successfully at least some of the time... but the motors are still getting very hot.
I'll have another twiddle at the pots, but I thought I'd mention, in case anyone has any other cunning ideas.

Hairyloon, 2020-05-08:
I have come to the opinion that the first driver board is duff, and trying to adjust that threw me off key for the others.
And possibly these motors are not as high rated as I had thought.
But it seems to be better now.
Thank you all for your help.

closed 2020-05-08 by Hairyloon