Klipper3d/klipper #2847: Bad Section for instantaneous corner velocity

theshark81, 2020-05-08:
Hi ,

I'm using Klipper in version : 

and try to add the instantaneous corner velocity to the extruder however i'm stuck because each time the klipper config is incorrect: the error is : 

[extruder]
step_pin = P2.13
dir_pin = P0.11
enable_pin = !P2.12
step_distance = 0.002349425
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = P2.7
sensor_type = EPCOS 100K B57560G104F
sensor_pin = P0.24
control = pid
pid_kp = 16.07
pid_ki = 1.37
pid_kd = 46.97
min_temp = 0
max_temp = 280
pressure_advance = 0.20
max_extrude_only_accel = 700
max_extrude_only_distance = 100.0

theshark81, 2020-05-08:
Here it is
[klippy.log](https://github.com/KevinOConnor/klipper/files/4599240/klippy.log)

Hywelmartin, 2020-05-08:
21 of Nov  2019

theshark81, 2020-05-08:
Thanks , i'll flash the firmware then mistake from my side !

closed 2020-05-24 by KevinOConnor