Klipper3d/klipper #2848: Problem with "BL touch failed to raise probe"

WizardOfYendor1, 2020-05-08:
I've been using Klipper on my Ender 3 for months now without issue. Today I tried to start a print and when it attempted to home the Z Axis I get a "BLTouch failed to raise probe" error.  

The probe does actually raise (and lower). In fact I can submit the debug commands BLTOUCH_DEBUG COMMAND=pin_down & pin_up and the probe moves as expected.... When I run QUERY_PROBE, it reports "triggered" when it is down AND when it is up. 

I should note I've changed nothing... literally nothing. Just turned it on to print something like I've done for the past few months and this just started happening.  I've tried various power cycling/reboots etc... without success. 

Attached is the log file.   Appreciate any insight/help. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4600049/klippy.log)

WizardOfYendor1, 2020-05-08:
And just like that..... I tried again just now and it works fine... didn't even power cycle since the last failure. 

Intermittent problems are the best :-) ;  guess we can close this out unless you have some suggestions.

pusewicz, 2020-05-08:
Just yesterday was having the same problem. Oddly enough, when doing 5x5 mesh grid, it would always fail on the 24th point. I'd repeat the mesh grid 10 times and it would always fail on the same spot. And then, just like that it would stop failing, even with multiple repeated bed meshes. 🤷

WizardOfYendor1, 2020-05-08:
Hmmm... if it matters at all, the way I do the bed levelling is I have a macro that stores it in the micro;  then I just "load it" via gcode at every print. So every once in awhile I'll run the auto leveling macro. 

I didn't bed level before this print that starting throwing this error (haven't auto bed levelled in a while). 

To troubleshoot, I started running the auto bed leveller macro and it continued to fail with the same error..... then just did z-axis homing and it failed there too....  then I just left it for a few hours and then tried to home the z axis again for fun... then it worked :-). 

It smells like a hardware problem.... but not sure yet.

fxbisto, 2020-05-10:
I've seen similar deploy / retract issues on 5 different printers (using different firmwares & versions, including Marlin v1 & 2 ) and all of them were using a clone - usually labelled 3Dtouch.
Are you guys using 'genuine' ones? Even with the slightly more expensive Trianglelabs ones it seems the 3Dtouch can work for a while then start messing about.
Power-cycling the entire printer seems to help, perhaps there's some kind of static build up affecting the hall sensor inside.

My feeling is it's unlikely to be a firmware issue, especially if it's been running fine for ages and you haven't changed or updated anything?

pusewicz, 2020-05-10:
I'm using the genuine BLTouch.

KevinOConnor, 2020-05-11:
FYI, I recommend updating to the latest code as several improvements were made to Klipper's BL-Touch error recovery since your version.

-Kevin

taker218, 2020-05-11:
I have the same problem on my CR10.

Just finished a print and wanted to start another and now I always get the error "Failed to verify BLTouch probe is raised; retrying."

The QUERY_ENDSTOPS is always triggered, if the pin is up or down.

I also just updated to the latest version of Klipper with git pull and a restart of the service. Even powercycled everything, Raspberry and printer with no luck.

Do I need to flash a new firmware as well? Normally it should tell me to flash the new firmware, or not?
[klippy.log](https://github.com/KevinOConnor/klipper/files/4611647/klippy.log)

taker218, 2020-05-11:
Nevermind my problem. I just checked the wiring and it seems like the black/white wire just tore apart (don't really know how or why).

KevinOConnor, 2020-05-24:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-05-24 by KevinOConnor