Klipper3d/klipper #2850: [Feature Request] Macro command template variables for current system homing/mesh/tilt calibration

matthock, 2020-05-08:
It would be helpful to have variables for the macro system that indicate current printer calibration. Things like knowing if the system is already homed or if there's already a bed tilt or bed mesh profile active. This would then allow macros to skip these (sometimes long in the case of mesh leveling) during print startup if they're already set and valid, but still run them if the system has been power cycled or timed out and shut down the motors since the last print. Knowing if there's an active mesh would be particularly useful, since it is by far the longest of these, although knowing if any of them were set would allow retaining manually tweaked Z offsets as well.

dushyantahuja, 2020-05-09:
I think you'll find most of the variables in this document:

https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md

Also, if you look at the individual modules that you're interested in - they should have a get_status function, that provides data back to klipper. e.g. if you look at output_pin.py

https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/output_pin.py#L42

The get_status function provides the "value" of the pin.

KevinOConnor, 2020-05-24:
I agree that exporting more state is something that would be useful.  Interested developers will need to do that work and submit it.

-Kevin

closed 2020-06-14 by KevinOConnor