Klipper3d/klipper #2851: Printer instance has no attribute 'try_load_module'

markwinger, 2020-05-09:
I have been running klipper for quite a while.  I was trying to install dwc2 to no avail and now killer wont start.  I get the error:
Internal error during connect: Printer instance has no attribute 'try_load_module'

in the clipper terminal but I have no idea what this means.
Logs attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4603476/klippy.log)

Hywelmartin, 2020-05-09:
This is not an Klipper issue ... 

Try here https://github.com/Stephan3/dwc2-for-klipper

markwinger, 2020-05-09:
This is getting ugly.  I installed from that page and find this fails:

wget https://github.com/chrishamm/DuetWebControl/releases/download/2.1.6/DuetWebControl-SBC.zip

There is no 2.1.6 apparently.  So I installed using:

wget https://github.com/chrishamm/DuetWebControl/releases/download/2.0.0-RC5/DuetWebControl.zip

The install seems to work but after installing, dwc2 does not connect and octoprint does not work with klipper complaining about the "try_load_module"
Can I get 2.1.6 somewhere to see if that works?

markwinger, 2020-05-09:
Oh, and I tried this too:
https://github.com/manu7irl/klipper-DWC2-installer

but nothing works after installing that way.

Hywelmartin, 2020-05-09:
This is not a Klipper issue... DWC2 is not a part of Klipper... If you have any problems with pristine Klipper code then it's a Klipper issue

markwinger, 2020-05-09:
Sorry, I thought I was talking to the dwc2 guys.

mental405, 2020-05-13:
The issue is that this commit https://github.com/KevinOConnor/klipper/commit/787ed452c2abe3bad81ffaf732b0b2609e3534bb changed the name of the method that DWC2 is trying to execute. Since DWC2 is not a "main" klipper module ie: not maintained through this repo, it doesn't always get updated. You will either need to make the changes in the DWC python code yourself or wait for Stephan to push a commit with the updated references.

Kszajga, 2020-05-19:
> Sorry, I thought I was talking to the dwc2 guys.

Before you start Klipper, you should change /klipper/klippy/extras/web_dwc2.py line:49 to
self.stepper_enable = self.printer.load_object(config, "stepper_enable") as @mental405 mentioned. Now it is working for me perfectly :)

KevinOConnor, 2020-05-24:
As indicated, this github issue tracker is exclusively for use with code from this github repository.  Any questions about dwc2 would need to be forwarded to the dwc2 developers.

-Kevin

closed 2020-05-24 by KevinOConnor