Klipper3d/klipper #2852: [Question] CoreXYU/IDEX Kinematics

bisongde, 2020-05-10:
Hey there,

I'm currently working on a CoreXYU build that I'd like to use Klipper with. I saw the answer in #1401 about a year ago from Kevin that this setup would take modifications to the kinematics in corexy.py. 

It occurs to me that IDEX is already implemented in Klipper; and this is essentially just IDEX, but driving the second X carriage with a second matched set of motors. The CoreXY kinematics aren't really changed at all, both stages function as a normal CoreXY setup would, they're just each devoted to a single carriage.

With that in mind:

- What if anything would need to be changed to accomplish this? 
- What is the assumption with the IDEX example config provided? Normal cartesian kinematics with a dedicated stepper for the second carriage?
- If a dedicated stepper can be defined for a second carriage in IDEX for a cartesian design, it seems like the same should be possible with CoreXY as well. How would that be done?

Would be happy to submit a PR if I can be pointed in the right direction on what needs to be done.

Thanks!

KevinOConnor, 2020-05-11:
This looks like a dup of #1401 .  An interested developer would need to implement it.  (I, personally, don't know what needs to be done - that's a major part of what an "interested developer" needs to accomplish.)

-Kevin

KevinOConnor, 2020-05-24:
I'm going to close this in favor of the existing #1401 .

-Kevin

closed 2020-05-24 by KevinOConnor

vtmue, 2020-06-01:
> * What if anything would need to be changed to accomplish this?

> An interested developer would need to implement it.

Am in a similar situation, only that it'll probably take me substantially longer to complete my hw build due to very specific requirements, and complicated sourcing.
Expect to add 2nd extruder this autumn/winter. So if noone has implemented it by then, I'll probably dive into coding it.
Cheerio,
tobi

1walkerit, 2020-07-18:
Hi,
I am interested about COREXY IDEX too. May be I can give some kinematics rules I think it needs.
Notes on Kinematics:
The X motion of  printer follows the same motion kinematics as a standard coreXY
The Y motion of  printer follows a coreXY kinematics, but the b-motor is mapped to the c-motor
The U motion of  printer follows Cartesian coordinates, with only the c-motor driven
[image]

So is it possible to implement it? I am not possible to do it myself. (the g-code is for RepRap FW)