Klipper3d/klipper #2853: 'mcu' shutdown after restart

Taudris, 2020-05-10:
I sometimes get this error after sending `SAVE_CONFIG`:

    'mcu' shutdown: Missed scheduling of next pwm event

But not at any other time. (Maybe it has happened when using the restart host button in OctoPrint/OctoKlippy? I don't use it very often.) It can print for hours (using virtual SD or not) and be totally fine, so I'm not inclined to believe that it's a hardware problem.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4605205/klippy.log)

KevinOConnor, 2020-05-24:
Unfortunately, it doesn't look like the log file is complete (it isn't showing the software version information).

The error itself was caused by the host software not turning off the bed heater prior to performing an internal restart.  The micro-controller detects this, turns off the bed heater directly, and transitions into an error state.  It's unclear why the host software did not turn off the heater itself (the code is written to do that).

-Kevin

Taudris, 2020-05-24:
I updated to the latest version a few days ago and haven't seen the issue since. I'll reopen and add more diagnostics if it happens again.

closed 2020-05-24 by Taudris