Klipper3d/klipper #2854: Unknown pin chip name 'probe'

Mattyboyzhere, 2020-05-10:
Trying to install the use of a genuine BL touch with the use of the Creality kit, unfortunately,
[klippy.log](https://github.com/KevinOConnor/klipper/files/4606616/klippy.log)
 I can't see to get the sensor to work as I keep getting error codes when changing `endstop_pin: probe:z_virtual_endstop` 

Attached if my Klipper reports.

KevinOConnor, 2020-05-11:
You are using a very old version of Klipper - you should upgrade and retry (see the FAQ for upgrade instructions).

-Kevin

closed 2020-05-24 by KevinOConnor

maviles798, 2021-07-07:
i have the same issue i have the lastes update of the klipper, im using a bl touch 3.1 on an ender 6. What can i do?

maviles798, 2021-07-07:
here is a copy of printer .cfg
[printer.txt](https://github.com/KevinOConnor/klipper/files/6779299/printer.txt)

maviles798, 2021-07-07:
here is the log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/6779449/klippy.log)

nickelphil, 2021-07-25:
I had the same problem and I should preface this by saying that I am low level of understanding but for mine I hadn't commented out the lines you were supposed to and added the ones you are supposed to for bltouch yet when I flashed my firmware. I saved the changes to the config file and flashed the firmware again using FLASH_DEVICE=1209:beba and I could connect after that.