Klipper3d/klipper #2855: Mesh bed leveling with piezo sensors while using a virtual Z endstop is very difficult to configure

Taudris, 2020-05-10:
I'm trying to use mesh bed leveling and virtual Z endstop with an under-bed piezo setup, but I'm having issues with getting the configuration to account for the activation distance (how far the bed must move after coming into contact with the nozzle to trigger the piezo sensors). The activation distance for my setup seems to be very substantial at 0.1mm (I'm not confident in this particular value yet, but the final number should be close), so it must be accounted for in order to make this sensing setup usable.

I found that the `[probe] z_offset` applies to both mesh measurement and homing, so if you try to use it to account for the activation distance, the end result is an effective offset of 0, thus failing to account for the activation distance and causing problems.

I tried to use `[stepper_z] position_endstop` to account for the activation distance without affecting the mesh measurement, but it has no effect at all when using `probe:z_virtual_endstop`. [The documentation for `[probe]`](https://github.com/KevinOConnor/klipper/blob/787ed452c2abe3bad81ffaf732b0b2609e3534bb/config/example-extras.cfg#L552) simply says "do not define `position_endstop`", not that it will be silently ignored if you try to use it anyway.

The workaround I came up with finally is to set `[probe] z_offset` to what I think the activation distance is, then use `BED_MESH_CALIBRATE Z_OFFSET=0` to acquire a mesh and force it to not incorporate the configured activation distance.

At a minimum, the documentation should say something about this to help anyone else who might have this issue in the future, and Klipper should throw an error when trying to use `position_endstop` together with `probe:z_virtual_endstop`.

Even better would be to use `[stepper_z] position_endstop` when using a virtual Z endstop (and continue to also use the `[probe] z_offset`). Piezo users could leave the `[probe] z_offset` at 0 and set `[stepper_z] position_endstop` to the activation distance, and users of other sensors could leave `[stepper_z] position_endstop` at 0 and set `[probe] z_offset` (or use both, though I can't think of a practical reason to do so). Because leaving `position_endstop` set when using `probe:z_virtual_endstop` has no negative consequences, this new behavior would need to be documented as a config change to alert users who may have accidentally left `position_endstop` set when migrating to a virtual Z endstop.

My latest [klippy.log](https://github.com/KevinOConnor/klipper/files/4606707/klippy.log) trimmed, mostly here to shut up the bot 😛

Arksine, 2020-05-10:
There is no "Z_OFFSET" parameter for BED_MESH_CALIBRATE,  so that command should have no additional affect on the calibration procedure.

With regard to your initial observation, it seems to me that if your activation distance is .1 (ie you have a negative z_offset) then subtracting it from the mesh should be the correct procedure.  Without doing so the mesh will be relative to Z coordinate lower than 0.  If your z_offset is -.1 and the mesh probe's Z=-.1 at (150,150), that location should be Z=0 in the mesh, as no adjustment should be done there.

Taudris, 2020-05-11:
Welp, I'm not sure what I was doing wrong before, but it seems you are right, and the attempted argument to `BED_MESH_CALIBRATE` does nothing. I tried setting `[probe] z_offset` to 1mm, and saw the correct behavior that I would only have expected with the workaround I thought I had found (the nozzle floating at Z=0 even after a new mesh calibration was applied). Never mind me.

closed 2020-05-11 by Taudris