Klipper3d/klipper #2856: Unknown command:"T0" on extrusion

felixmendoline, 2020-05-11:
Hi,

i try to setup Klipper on a CR10S-Pro (v1) with a E3D Hemera Direct Kit and a custom build Carbon 750W Heatbad. So my Printer.cfg is a bit different then on a stock CR10S-Pro. 

I actually have 2 problems:
1) When I heat this up hot, it exceeds the specified temperature. But here i will try to see how it is after temp calibration.

2) If I want to extrude fillament, then I can only do this once. If I get the error xxx and have to restart the FW before I can send any command.

I have included a screenshot, the config and the Klippy.log in the appendix.

Thanks a lot,

Felix

[klipper.zip](https://github.com/KevinOConnor/klipper/files/4608592/klipper.zip)

BlackStump, 2020-05-11:
You need to change the behavior of Octoprint to not issue M112 on a unknown command, I would suggest you need to setup Octoprint to work with Klipper.
Also look at the config changes
[20191210](https://www.klipper3d.org/Config_Changes.html)

felixmendoline, 2020-05-11:
Ok, you was right. Somehow the Octoprint behaviour setting was not set as written in the documentation and I have had no config section to deal with T0. But the problem is more or less the same. I can only send one time a extrude command, the second time i got no reaction. OK now i do not get the failure (Unknown command:"T0") it simply do not happen anything. I attched a fresh klippy.log and also the updated config. Would be great if you could help me out.

As reference for others:
_

1.  _In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save"._
2. 20191210: The builtin T0, T1, T2, … commands have been removed. The extruder activate_gcode and deactivate_gcode config options have been removed. If these commands (and scripts) are needed then define individual [gcode_macro T0] style macros that call the ACTIVATE_EXTRUDER command. See the config/sample-idex.cfg and sample-multi-extruder.cfg files for examples.

_
[cr10s-pro hamera log v2.zip](https://github.com/KevinOConnor/klipper/files/4611731/cr10s-pro.hamera.log.v2.zip)

felixmendoline, 2020-05-16:
Hi, has anyone any idea?

BlackStump, 2020-05-16:
[try updating to the latest klipper](https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software)
My guess is it is still a Octoprint issue

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the original question was answered (it does not appear to be a Klipper issue).

-Kevin

closed 2020-06-14 by KevinOConnor

nighonimpossible, 2022-01-24:
@jel111 I stumbled across this old thread when searching for the "Unknown command T0" when using my Voron 0. After a little digging I can see the T0 code is actually being generated by the slicer (Cura in my case). If I believe the comments in the gcode file,  the T0 command is "Generated with Cura_SteamEngine master". I just need to figure out that Cura is doing and why.