Klipper3d/klipper #2857: Z_tilt Correction is reversed?

Moonlight63, 2020-05-12:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4613339/klippy.log)

I must have a fundamental misunderstanding of how the z_tilt function works. I am in the middle of building a corexy printer with a moving bed. The bed has 3 belt drive axis and is mounted with 3 pivot points. Like so:
```
  -----------
  |         Z1
  |         |
  Z0        |
  |         |
  |         Z2
  -----------
0,0
```

The pivot positions look something like this:

```
  -----------
  |      380,335
  |         |
95,180      |
  |         |
  |      380,25
  -----------
0,0
```

And the probe points are just inside of this grid. My z_tilt section looks like this:

```
[z_tilt]
z_positions:
    95, 180
    380, 335
    380, 25
points:
    100, 180
    350, 300
    350, 100
```

So here is my problem. When I run the Z_TILT_ADJUST code, everything works fine, until the bed goes to level. The motors moved to compensate, but in the wrong direction I guess? The easiest way to show this is to just look at the numbers. So I'll do to Z_TILT_ADJUST commands in a row, one after the other with nothing in between. Text below pulled directly from klippy.log

First run:
```
probe at 100.000,180.000 is z=-0.694981
probe at 350.000,300.000 is z=1.052124
probe at 350.000,100.000 is z=0.328185
Calculating bed tilt with: [[99.99999999978886, 179.99999999979278, -0.69498072000642], [350.0000000004633, 300.0000000004644, 1.0521235899871542], [350.0000000000165, 100.0000000001836, 0.3281853399807346]]
Coordinate descent initial error: 1.69766786717
Coordinate descent best_err: 6.24119025254e-11  rounds: 232
Calculated bed tilt parameters: {'y_adjust': 0.0036196601588156445, 'x_adjust': 0.005250965942561332, 'z_adjust': -1.8716195942421296}
Making the following Z adjustments:
stepper_z = -0.721239
stepper_z1 = 1.336334
stepper_z2 = 0.214239
```

Second run:
```
probe at 100.000,180.000 is z=-0.919689
probe at 350.000,300.000 is z=-0.031659
probe at 350.000,100.000 is z=0.547492
Calculating bed tilt with: [[99.99999999978886, 179.99999999979278, -0.9196894988876423], [350.0000000004633, 300.0000000004644, -0.03165857889406709], [350.0000000000165, 100.0000000001836, 0.547492021099508]]
Coordinate descent initial error: 1.14657855315
Coordinate descent best_err: 7.87339671206e-11  rounds: 220
Calculated bed tilt parameters: {'y_adjust': -0.0028957654126841267, 'x_adjust': 0.004942068379473075, 'z_adjust': -0.8926601731206323}
Making the following Z adjustments:
stepper_z = -0.944401
stepper_z1 = 0.015244
stepper_z2 = 0.912932
```

Third run:
```
probe at 100.000,180.000 is z=-1.787261
probe at 350.000,300.000 is z=-0.069114
probe at 350.000,100.000 is z=1.147102
Calculating bed tilt with: [[99.99999999978886, 179.99999999979278, -1.787260678645687], [350.0000000004633, 300.0000000004644, -0.06911389865211248], [350.0000000000165, 100.0000000001836, 1.1471023613414602]]
Coordinate descent initial error: 4.51492129182
Coordinate descent best_err: 1.13961670872e-10  rounds: 232
Calculated bed tilt parameters: {'y_adjust': -0.006081073395547153, 'x_adjust': 0.009791536179198436, 'z_adjust': -1.671831147205059}
Making the following Z adjustments:
stepper_z = -1.836228
stepper_z1 = 0.011793
stepper_z2 = 1.896926
```

The deviation gets worse with every subsequent run. The best I can figure is that the Z_tilt compensation is going the opposite direction that it should. This is most apparent with the main stepper_z motor. Instead of the offset improving, it's getting worse. This is easy to tell visibly because that motor is responsible for the bed tilt in the x direction based on the crappy ASCII drawing above.

Please help me? im confused...

test3210-d, 2020-05-13:
there are doubts about the correctness of the points of attachment of the belt relative to the zero point . It is critically important. 

incorreect motor position x = 284 mm
`Send: z_tilt_adjust
Recv: // probe at -9.000,150.000 is z=0.762500
Recv: // probe at 230.000,150.000 is z=0.722500
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.113331
Recv: // stepper_z1 = -0.054431
Recv: // Retries: 0/4 Probed points range: 0.040000 tolerance: 0.030000
Recv: // probe at -9.000,150.000 is z=0.798331
Recv: // probe at 230.000,150.000 is z=0.953331
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.166600
Recv: // stepper_z1 = -0.394867
Recv: // Retries: 1/4 Probed points range: 0.155000 tolerance: 0.030000
Recv: // probe at -9.000,150.000 is z=1.182367
Recv: // probe at 230.000,150.000 is z=1.004867
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.178955
Recv: // stepper_z1 = 0.082458
Recv: !! Retries aborting: Probed points range is increasing.`

approximattily correct value x=286 mm 

`Send: z_tilt_adjust
Recv: // probe at -9.000,150.000 is z=0.690000
Recv: // probe at 230.000,150.000 is z=0.787500
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.149814
Recv: // stepper_z1 = -0.294242
Recv: // Retries: 0/4 Probed points range: 0.097500 tolerance: 0.030000
Recv: // probe at -9.000,150.000 is z=0.944242
Recv: // probe at 230.000,150.000 is z=0.894242
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -0.230761
Recv: // stepper_z1 = -0.156715
Recv: // Retries: 1/4 Probed points range: 0.050000 tolerance: 0.030000
Recv: // probe at -9.000,150.000 is z=1.210761
Recv: // probe at 230.000,150.000 is z=1.240761
Recv: // Making the following Z adjustments:
Recv: // stepper_z = 0.066864
Recv: // stepper_z1 = 0.022449
Recv: // Retries: 2/4 Probed points range: 0.030000 tolerance: 0.030000
Recv: ok``

Moonlight63, 2020-05-14:
Ok, so I have made some changes that seem to have helped quite a bit. I moved the points around slightly to make them more accurate. They are now dead on with what my printer cad model says they should be. I have also enabled retries which gives me more information. I now have a situation very similar to the example you have posted. The first adjustment corrects a large amount of the bed tilt (by default, the bed starts out pretty badly tilted do to a weak motor that I plan to replace later, causing one side to fall, by in this case is helping me to figure out this z_tilt setup), subsequent retries then cause the calculated error to rise on each run. Based on your suggestion, it sounds like the issue is that I down have the pivot points set exactly to where they should be. I am using a micro switch as my z probe ( I am about to run a repeat-ability check on it now just to rule that out ), and there is no toolhead mounted currently. I am building/have built a toolchanger that works great, and the plan is to simply use this micro switch as the z probe when no tool is mounted, then set a nozzle offset for each tool. So now my question is, what exactly should I be setting the z_positions and points options relative to? Currently, I have them set relative to roughly the center of the switch, but maybe that's not good enough? It is one of the switches with an arm on it. Do you have any suggestions on how to dial in the true location of these pivot points?

test3210-d, 2020-05-14:
z_positions  - relative to extruner nozzle center 
if swich offset in yor case = 0,0 all must ork fine.

Moonlight63, 2020-05-14:
OK, I am supper confused now. I made a video showing what is happening. Just as a sanity check I made one of the probe points touch off on the bed corner mount, which is high. The bltouch is set to 0,0,0 offset. I verified the position of the pivot points by lining up the probe directly above them. By my understanding, if the probe touches off on a corner that is high, either that corner must lower, or the other corners must rise to compensate. I would have to imagine that even if my pivot points are off by half a mm that this would still mostly work for an initial correction. Instead, literally the complete opposite happens. Here is the video showing this: https://www.youtube.com/watch?v=b73lzE822_4

test3210-d, 2020-05-15:
swap motors and check, i think something wrong in config or hardware

Moonlight63, 2020-05-15:
Swap them how? Like change the order? What order should they be in?

closed 2020-05-15 by Moonlight63