Klipper3d/klipper #2858: Clock selection when setting compile parameters for STM32F407

Lefuneste83, 2020-05-12:
Hi there.

Just a quick question not an issue here.

When running make menuconfig for an SKR Pro V1.1 (STM32F407 MCU) I have the option to select various clock speeds.

I believe from my last run than you can choose 8 MHz, 12 MHz, 16 MHz and Internal Clock. I have been searching around to find some explanations on the impact this setting would have but haven't found a clear answer. To complete my understanding of this fabulous piece of software that Klipper is, what is the purpose of this setting ? By default I've selected 8 MHz and the compiled firmware seems to be working. Are there any reasons I would want to tinker with this setting ? Thanks

Arksine, 2020-05-12:
That setting needs to match the crystal oscillator connected to the MCU.  In the SKR Pro's case the crystal is 8Mhz.  If this setting is not correct the most likely scenario is that the peripherals will not work correctly (such as USB), making the board temporarily unusable until you flash it with the correct setting.

Lefuneste83, 2020-05-13:
Thanks for your reply. I was wondering if there would be some benefit in tweaking this setting but obviously not... I'll stick with 8MHz then...

closed 2020-05-24 by KevinOConnor