Klipper3d/klipper #2859: filament_switch_sensor only detects load event

bistory, 2020-05-12:
Hi !

First thanks for Klipper, I use it for more than one year and it's really a pleasure every time I need to tinker with my printer.

Recently, I tried to use the filament_switch_sensor feature as I own an Anycubic i3 Mega which has such a detector that I didin't use anymore as it was not supported by Klipper.

However, it doesn't seem to work correctly as only the load event is detected. When filament is unloaded, everything continues like nothing happened...

Here are the relevant part of my config and the log :

`[pause_resume]
recover_velocity: 60.
#   When capture/restore is enabled, the speed at which to return to
#   the captured position (in mm/s).  Default is 50.0 mm/s.

# Filament Switch Sensor.  Support for filament insert and runout detection
# using a switch sensor, such as an endstop switch.
[filament_switch_sensor filament_sensor]
pause_on_runout: True
#   When set to True, a PAUSE will execute immediately after a runout
#   is detected. Note that if pause_on_runout is False and the
#   runout_gcode is omitted then runout detection is disabled. Default
#   is True.
runout_gcode:
    M118 Filament Runout Detected
#   A list of G-Code commands to execute after a filament runout is
#   detected. See docs/Command_Templates.md for G-Code format. If
#   pause_on_runout is set to True this G-Code will run after the
#   PAUSE is complete. The default is not to run any G-Code commands.
insert_gcode:
    M118 Filament Load Detected
#   A list of G-Code commands to execute after a filament insert is
#   detected. See docs/Command_Templates.md for G-Code format. The
#   default is not to run any G-Code commands, which disables insert
#   detection.
#event_delay: 3.0
#   The minimum amount of time in seconds to delay between events.
#   Events triggered during this time period will be silently
#   ignored. The default is 3 seconds.
#pause_delay: 0.5
#   The amount of time to delay, in seconds, between the pause command
#   dispatch and execution of the runout_gcode.  It may be useful to
#   increase this delay if Octoprint exhibits strange pause behavior.
#   Default is 0.5 seconds.
switch_pin: ^!ar19
#   The pin on which the switch is connected. This parameter must be
#   provided.`

[klipper.log](https://github.com/KevinOConnor/klipper/files/4615371/klipper.log)

Thanks !

manu7irl, 2020-05-12:
So the normal behavior is reversed?

bistory, 2020-05-12:
Yes it seems to be, when I remove the reversed flag, it triggers filament load when I remove the filament from the sensor ;)

Arksine, 2020-05-12:
The runout event will only fire while the printer is in a "printing" state.  Try inserting the filament, wait a few seconds, then run the following gcode command:
```
G4 P100000
```
That will provide you 10 seconds where the printer is in the printing state (it wont actually do anything but "wait").  If you pull out the filament during that time, the printer will run the PAUSE gcode then the execute runout gcode.   After doing this be sure to run a CLEAR_PAUSE or restart the printer to make sure that the paused state is cleared.

bistory, 2020-05-12:
I feel dumb but you are right @Arksine, it was the root of my problem ! Thanks ! I can close this :-)

closed 2020-05-12 by bistory