Klipper3d/klipper #2862: Unable to Connect

SmarterTec, 2020-05-13:
Klipper was running fine for the past month or so but I just changed my nozzle and for some reason I can no longer connect. 

I used a macro I have used numerous times before which unloads my filament. After I unloaded the filament, I changed the nozzle and went to use the reload filament macro but noticed my LCD was frozen.

I tried restarting firmware, restarting octoprint, rebooting the system, and unplugging and plugging back in but keep getting this error message:

04:17:22 Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Klipper state: Not ready
04:17:22 Unable to connect

Everything was working perfectly and I didn't make any software changes since my last print so not sure what could have changed. 

Any ideas what could have caused this?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4619632/klippy.log)

BlackStump, 2020-05-13:
Check your USB Cable and your wiring. Does the SKR Mini E3 have any leds on the board that are lit and have you checked you are getting power to the SKR

mental405, 2020-05-13:
In your log file you have
`Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD9054242353929732157-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD9054242353929732157-if00'
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 135, in _connect
    self.send_event("klippy:mcu_identify")
  File "/home/pi/klipper/klippy/klippy.py", line 207, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/mcu.py", line 594, in _mcu_identify
    raise error(str(e))
error: Unable to connect`

This typically signifies that the printer board is not responding via the serial port for any number of reasons. If you recently changed nozzles, there is a possibility that you shorted the heater to the wrench and fried your board.

SmarterTec, 2020-05-16:
The board does have a red LED which is lit. I switched out the USB cable but am still having the same issue. I will order a replacement board and see if that fixes it. Thanks

SmarterTec, 2020-05-25:
Okay, I purchased a new SKR mini e3 v1.2 and installed it. Confirmed that the printer is working with Marlin 2.0.

I re-flashed the board with the firmware.bin file I used to flash originally. I changed the mcu serial to the one reported by the new board which is /dev/serial/by-id/usb-LeafLabs_Maple-if00

Tried connecting again but still getting the unable to connect error. I've attached an updated klippy.log. Not sure if maybe I missed a step when reflashing the board which could be causing the issue? 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4678166/klippy.log)

SmarterTec, 2020-05-25:
Update,  for some reason the mcu serial was incorrect. I reran the command and got a different value. Confirmed it's working now after changing the serial in the config file.

closed 2020-05-25 by SmarterTec

Jubidition, 2021-03-12:
@SmarterTec any chance you can share what you did to get the right info and up running. I am having the same exact issue at the moment and can't get Klipper to run on the BTT SKR mini E3 V1.2 since it will not connect to the MCU - I hope you can help me out

mental405, 2021-03-12:
This issue was closed almost a year ago so the likelihood of getting a useful response is low.

Additionally, it will also not trigger the bot, so I will copy and paste the bot response.

```
Unfortunately, it appears the directions at https://www.klipper3d.org/Contact.html were not followed for this issue.

We choose to use github as a place that people working on improving Klipper can share the results of their work. We choose to not use github to help diagnose problems with a user's printer. We choose to not use github to answer user questions. We choose to not use github as a place to make requests.

In the future, please follow the directions at: https://www.klipper3d.org/Contact.html
This ticket will be automatically closed.

Best regards,
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.
```

SmarterTec, 2021-03-13:
@Jubidition  My board was fried and I had to order a new one. It shorted when I changed the nozzle. 
After I replaced it, I followed the normal Klipper install procedure. I just messed up one step and input the wrong serial id but if you follow the guide you shouldn't have the same issue.