Klipper3d/klipper #2863: [REQUEST] check klipper with nextion connected to the raspberry

b1k386, 2020-05-13:
I am very fascinated by klipper, before starting the conversion of my printer I was wondering if it was possible to control klipper also through nextion connected to the raspberry which in turn is connected to the printer board, unfortunately in addition to remote control I also need a local control through the nextion connected to the raspberry

jakep82, 2020-05-13:
This question would be better aimed at Octoprint.  If the nextion works with Octoprint, it can control Klipper.  I see some touchscreen interfaces for Octoprint, but I've never used one so I can't say how well they work, or if the nextion is compatible.

KevinOConnor, 2020-05-24:
FWIW, I'm unfamiliar with "nextion" and thus I can't personally help with this question.

Cheers,
-Kevin

closed 2020-06-14 by KevinOConnor