Klipper3d/klipper #2864: Question: Is there a way to set auto-start gcode?

dushyantahuja, 2020-05-13:
Smoothieware has an option where you can create a on_boot.gcode file on the SD card, and it runs the file every time it boots. Is there a similar option in Klipper - so that set gcode runs automatically whenever the printer is connected?

mental405, 2020-05-13:
Depends on your definition of auto start gcode.

You can add start gcode to the octoprint macros when a print is started.
You can create a delayed gcode [macro] (https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L434)that executes after a certain period of time when the printer enters the "ready" state
You can add start gcode to your slicer profile

dushyantahuja, 2020-05-13:
Thanks a lot, didn't realise that delayed gcode could be used for this.

closed 2020-05-13 by dushyantahuja