Klipper3d/klipper #2865: ABL Z axis inverted. Goes up when should be going down.  Inductive probe.

ChrisMLian, 2020-05-13:
I have a ender 3 with octoprint and klipper (same issue in merlin) firmware and an inductive probe. For years ive had this issue with the bed level compensation being reverse. When it should be going down it goes up. Eventually i just gave up on it.  Its the same in Prusa and Cura.  I would be extremely grateful if this could be solved. 

https://www.ebay.com/itm/Anet-A8-LJ18A3-8-Z-BX-Inductive-Proximity-Sensor-3d-printer-upgrade/174273051977?hash=item28937bbd49:g:UdcAAOSw81Zesivz

Slicers are configured with 0,0 being in the lower left corner 
x   x  x 
x       x
0 x x x

Is there a way to compensate for this? 
Here's my klipper config file

[printer.txt](https://github.com/KevinOConnor/klipper/files/4621294/printer.txt)
[image]

ChrisMLian, 2020-05-13:
Apologies forgot to add it. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4621614/klippy.log)

Hywelmartin, 2020-05-13:
I don't use mesh.. but you have some saved meshes in a 3x3 format.. erase them and make new??

ChrisMLian, 2020-05-13:
> I don't use mesh.. but you have some saved meshes in a 3x3 format.. erase them and make new??

These are just saved meshes and not in use. It makes a new mesh and use that every time.  Its not related to the issue either way

Arksine, 2020-05-13:
The first thing you should do is verify that your [x and y offsets are accurate](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#calibrating-probe-x-and-y-offsets).  If they are off then the correction applied will be off.

After that I would recommend doing the [probe repeatablility](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#repeatability-check) and [location bias](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#location-bias-check) checks.   The issue you are experiencing sounds like location bias, which would indicate a mechanical issue such as a twisted x axis.

ChrisMLian, 2020-05-13:
> The first thing you should do is verify that your [x and y offsets are accurate](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#calibrating-probe-x-and-y-offsets). If [...]

Looks it could be The X axis or bias yes.  When i tilt the bed a fair bit to either side it does compensate the correct way

closed 2020-06-01 by KevinOConnor