Klipper3d/klipper #2866: neopixel wont work anymore after update to latest build

Harrdy, 2020-05-13:
With included Neopixel block, klipper are unable to start.

#[neopixel caselight]
#pin: PC4
#chain_count: 18
#initial_RED: 0.0
#initial_GREEN: 0.0
#initial_BLUE: 0.0
#color_order_GRB: True

FIRMWARE_VERSION:v0.8.0-526-gf0c394de

Protocol error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 559, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 525, in _send_config
    self._serial.send(c)
  File "/home/pi/klipper/klippy/serialhdl.py", line 181, in send
    cmd = self.msgparser.create_command(msg)
  File "/home/pi/klipper/klippy/msgproto.py", line 328, in create_command
    raise error("Unable to extract params from: %s" % (msgname,))
error: Unable to extract params from: config_neopixel
Build file /home/pi/klipper/klippy/../.config(1427): Wed May 13 19:18:51 2020

[klippy.log](https://github.com/KevinOConnor/klipper/files/4623693/klippy.log)

Harrdy, 2020-05-13:
klipper dont tell me to update the mcu. but i wanted to try it. after flashing the new mcu firmware and re-enabling the neopixel config part, everything is working again as if nothing had ever happened

mental405, 2020-05-13:
Commit https://github.com/KevinOConnor/klipper/commit/a7f7cc507547758879890b206030e508ac8dbdb0

added the ability to configure the timings from the host software which necessitated a code change to the micro controller code. This requires the micro controller code to be recompiled and flashed so that the timings are sent when the strips are configured rather than being hard coded in the MCU code.

KevinOConnor, 2020-05-21:
When upgrading the software, be sure to follow the steps at https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

-Kevin

closed 2020-06-14 by KevinOConnor