Klipper3d/klipper #2867: Nozzle Always prints directly on bed

Oniketsu, 2020-05-13:
Hello, I'm having an issue where no matter how I set my config, the first layer is always way too close to the bed, I have an Ender 3 Pro and a Bltouch, copy of my config and log attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/4624915/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4624900/printer.cfg.txt)

jakep82, 2020-05-14:
Have you calibrated your bltouch offsets?

https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md

Oniketsu, 2020-05-14:
I did and now I'm getting a BLTouch failed to deploy issue even though the calibration completed successfully and offset setting now seemed to actually do something
[klippy.log](https://github.com/KevinOConnor/klipper/files/4626096/klippy.log)

Oniketsu, 2020-05-14:
The BLTouch pin was bent, replacing the pin fixed it.

Thank you

closed 2020-05-14 by Oniketsu