Klipper3d/klipper #2868: Nozzle hitting bed after calibration

sshaikh, 2020-05-14:
I suspect this is a documentation/understanding issue rather than a technical one. I do not have logs at hand but will to update this issue when I have constructed an illustrative set.

After reviewing the documentation I calibrate my bed in the following order (with no probe):

1. BED_SCREWS_ADJUST
2. Z_STOP_CALIBRATE
3. BED_MESH_CALIBRATE (manually)

However when I print, the purge line hits the bed resulting in a lovely trough along the side of my bed after a few prints. Prints themselves are otherwise great.

**Observations:**

The eventual created mesh shows that my bed is bowed downward toward middle.

My screws are in the corners of the bed, and so BED_SCREWS_ADJUST brings the edges of the bed level to the nozzle (at z=0?). It is unclear if this is a calibrated zero due to previous Z_STOP_CALIBRATE or not.

Thus my Z_STOP_CALIBRATE performed towards the centre of the bed as per the docs is negative. When viewing the mesh I see positive values along the left and right edges and negative values along the middle axis. 

This all makes some sense to me, except for the nozzle hitting the bed during prints and perhaps even homing (I need to confirm). Previously when I omitted the BED_SCREWS_ADJUST step I didn't see the issue, so I suspect the negative zstop value is being applied at the edges of the bed.

**Questions:**

1) Shouldn't the mesh edges have zero numbers, as that is where the bed is set to via the screws?
2) Does the created mesh get loaded by klipper automatically during prints and homing? I don't recall having to do anything similar after manually creating the mesh in Marlin.
3) Should Z_STOP_CALIBRATE always be performed in middle of the bed or at a known highest point (ie the edges for my printer).

sshaikh, 2020-05-15:
Attached logs.

Since posting the issue I have had another fiddle, including physically moving the z-stop. I still calibrated in the same order and in the same places, however this time I ended with a positive value for z position_endstop.

I now see the z value follow the mesh as the nozzle moves across the XY plane, both in and out of prints, and my purge line doesn't appear to be hitting the bed anymore (although I will need to see if there is any further visible damage after a few more prints).

Interestingly this doesn't reflect in the printer display, which shows a value of 0 for Z after homing (and in fact I initially got the nozzle to hit the bed when moving it around the bed using the display).

So I think this has been solved, at least for me, by ensuring the z position_endstop is a positive value (ie by moving the endstop up). I think my confusion stemmed from a single line in Manual_Level.md:

> Some printers have the ability to manually adjust the location of the physical endstop switch. However, it's recommended to perform Z endstop positioning in software with Klipper

Which led me to believe I shouldn't touch my endstop at all. Now that I have a better understanding of how things work the context around the statement makes sense.

I've attached logs nonetheless just in case they're useful:

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4635389/klippy.zip)

KevinOConnor, 2020-05-24:
If using BED_SCREWS_ADJUST then one should not use Z_ENDSTOP_CALIBRATE.

The Z_ENDSTOP_CALIBRATE tool is used to calibrate the Z endstop relative to a fixed bed position.  In contrast, the BED_SCREWS_ADJUST tool is used to calibrate the bed position relative to a fixed Z endstop position.

So, you want to use the BED_SCREWS_ADJUST tool exclusively.

-Kevin

sshaikh, 2020-05-24:
Thank you for the elaboration. This makes sense, to make the z stop the absolute zero.

It might be useful to explain that only one of these calibrations is required depending on the bed. Also the bed mesh instructions specifically mention that z endstop be calibrated using Z_ENDSTOP_CALIBRATE.

I'm still a little puzzled by the way to approach a bed with negative curvature like mine, as if levelled at the corners then the middle of the bed will have a negative z position. My stop just about allows this so it's not a problem yet. In any case I guess it's not a klipper issue, but I guess you would need the ability to home Z at a custom XY location, as well as adjust screws at an arbitrary height.

closed 2020-06-14 by KevinOConnor