Klipper3d/klipper #2869: SET_GCODE_OFFSET переопределяется экструдерами. Нужна независимая переменная

Goodfeat, 2020-05-14:
Hello. How to use the variable SET_GCODE_OFFSET simultaneously for offset table and two extruders.

Using the SET_GCODE_OFFSET variable, I set the gap between the nozzle and the table and this works fine.

But if I use this option additionally to offset the extruder, then the set clearance above the bed no longer works. This creates inconvenience when changing the tool, since SET_GCODE_OFFSET is always redefined.

I have the following in my code:
```
[homing_override]
axes: xyz
gcode: SET_SERVO SERVO = myservo ANGLE = 10
       G28 Z0
       G91
       G1 Z-37 F1200
       G90
       SET_GCODE_OFFSET Z = 0.50

[gcode_macro T0]
gcode:
       SET_GCODE_OFFSET Z = 0 MOVE = 1 # Adjust z-height
       SET_GCODE_OFFSET X = 9.9 MOVE_SPEED = 100 # Clear X offset
       ACTIVATE_EXTRUDER EXTRUDER = extruder

[gcode_macro T1]
gcode:
       SET_GCODE_OFFSET Z = 5 MOVE = 1 # Adjust z-height
       SET_GCODE_OFFSET X = -9.9 MOVE_SPEED = 100 # Clear X offset
       ACTIVATE_EXTRUDER EXTRUDER = extruder
```

In this case, as soon as the first tool is selected, the total offset above the table will be redefined SET_GCODE_OFFSET Z = 0?

that is, in this code, adding SET_GCODE_OFFSET Z = 0.50 to the [homing_override] section makes no sense, since it will be immediately redefined by the T0 command

Is there a separate variable for table offset, or is there a way to set offset for an extruder without touching the SET_GCODE_OFFSET variable without additional calibrations?

I have a delta printer and switched from Repeti

er firmware. 
In the repeater firmware, there are separate variables for calibrating the gap between the table and the nozzle, as well as the offset of the extruders themselves relative to the center.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4626076/klippy.log)

KevinOConnor, 2020-05-24:
The typical way to configure the "distance between the nozzle and bed" after a homing operation is to use the [stepper_z] position_endstop variable.

If you do need to issue SET_GCODE_OFFSET during both the homing operation and the tool change operation then you would need to write those macros knowing that there might be an offset already in place.  If so, it's possible to query the current offsets and it is possible to store state in variables (see docs/Command_Templates.md for the details).

-Kevin

Goodfeat, 2020-05-25:
Unfortunately, on a delta printer, the setting is not so clear. 
Setting  endstops becomes difficult if the delta printer is automatically calibrated. In this case, it is necessary to change the parameters of all three motors. In the documentation In docs / Command_Templates.md, only the basic rules for using macros are prescribed.
There are no examples in the documentation of using SET_GCODE_OFFSET for both the table and the tools (two nozzles with offset).

Although I found a way out due to the flexibility of the Klipper. in this case, I use the save_variables.py script from @dushyantahuja. This script helped expand the functionality of the Klipper.

```
[gcode_macro T0]
gcode:
	{% if printer.toolhead.homed_axes == "xyz" %}	
		SET_GCODE_OFFSET X=9.9 Z={printer.save_variables.zoffsettool|float + printer.save_variables.zoffsetbed|float } MOVE=1 MOVE_SPEED=100 
		SAVE_GCODE_STATE NAME=EXTR0
		ACTIVATE_EXTRUDER EXTRUDER=extruder
		RESTORE_GCODE_STATE NAME=EXTR0
	{% endif `%}`

[gcode_macro T1]
gcode:
	{% if printer.toolhead.homed_axes == "xyz" %}	
		SET_GCODE_OFFSET X=-9 Z={printer.save_variables.zoffsettooltwo|float + printer.save_variables.zoffsetbed|float } MOVE=1 MOVE_SPEED=100 
		SAVE_GCODE_STATE NAME=EXTR1
		ACTIVATE_EXTRUDER EXTRUDER=extruder1
		RESTORE_GCODE_STATE NAME=EXTR1
	{% endif `%}`
```

In a separate configuration file, I specified offset for the tool and offset for the bed. In this way, you can adjust the offset of the table, and the nozzle will be positioned correctly

if for the first nozzle I have printer.save_variables.offset = 0, then for the second nozzle I have shifted not only along the X and Y axis, but also along the Z=5 (This is my printer configuration. The second tool switches along the Z axis)

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-06-14 by KevinOConnor