Klipper3d/klipper #2870: Unable to read tmc uart (SKR 1.3 / TMC 2208)

4noxx, 2020-05-14:
hello 

my board is a SKR 1.3 with TMC 2208 Driver.

I ve some problems with my configuration
Unable to read tmc uart .....

1) under my driver all jumpers are removed
2) red are set

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/4627275/klippy.log)

4noxx, 2020-05-14:
i change the uart pins in the config, message "Unable to read tmc uart ....." is gone, but printer will not work

[klippy.log](https://github.com/KevinOConnor/klipper/files/4627371/klippy.log)

dianlight, 2020-05-14:
Hi,
 I'have the same board but 2209 UART that are similar. Why you change the pins?

The correct config is: [generic-bigtreetech-skr-v1.3.cfg](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-v1.3.cfg)

Also many TMC2208 based boards require soldering to enable UART mode. Check the manufacturer's information.

As I can see modify your config to:
...
[tmc2208 stepper_x]
uart_pin =  ~~P4.29~~  P1.17
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250

[tmc2208 stepper_y]
uart_pin = ~~P1.16~~ P1.15
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250

[tmc2208 stepper_z]
uart_pin = ~~P1.14~~ P1.10
microsteps = 16
run_current = 0.650
hold_current = 0.450
stealthchop_threshold = 30

[tmc2208 extruder]
uart_pin = ~~P1.9~~ P1.8
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 5
....

vladT0, 2020-05-20:
Please see also https://github.com/KevinOConnor/klipper/issues/542

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the original question was answered (use correct pins and be aware of issue with some stepsticks).

-Kevin

closed 2020-06-14 by KevinOConnor