Klipper3d/klipper #2871: bed mesh first layer

tpipits, 2020-05-14:
Hi to all.

I use klipper 6 months now without a problem on a modified anycubic i3 mega.
Recently i add a probe at my printer (endstop microswitch with sg90 servo) because my bed was a mess (i think is y rods issue but when i fix their base after two or three prints the problem is back so that the reason i try to enable mesh bed leveling.
I use my z/z1 endstops because i want my x axis leveling at z homing.
I have probed my bed at 100 deg Celsius (ABS printing).  
The bed mesh seems fine i think correspond the mess of my bed.
Before the print start i run mesh_bed_clear and load default mesh and then i start the print.
BUT... at the left side the first layer is squeezed, at the middle is almost perfect and at the right side nozzle is too high and the layer lost adhension or not bond together.
I have tried recalibrating endstop position and probe z-offset with no differrence between left and right side (only the layer becomes at total more squeezed or not). 
Also i have tried use octoprint and DWC2 with no differece and update klipper to latest with no difference.
It seems as the x axis of bed med is inverted at printing.....
Please help i print only in the middle of the bed two weeks now and i can't find the problem.
I attach pictures of bed visualizer, cfg section of bed mesh level, firts layer of test cube print and my klippy log file.

Thank you for any response. 

[image]
[image]
[image]

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4627769/klippy.zip)

Arksine, 2020-05-14:
Read the [Probe Calibrate] document and make sure that your x and y offsets are absolutely correct.  Then do the probe repeatability check to to make sure that the probe is accurate.

If you are still experiencing the issue after you know the offsets are correct and the probe is accurate, do the Location Bias check.   Location bias would indicate a a flaw with the printer itself (such as a twisted X axis) that causes the probe to trigger too high on the right and too low on the left.

ArDrift, 2020-05-15:
It might not be the same with your probe, but I found that with mine, I experience serious [temperature bias](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#temperature-bias).

When probing the bed on operating temperatures (~60°C), the result is effectively the same as in your example, the bed mesh is not accurate. As also pointed out in the probe calibrate document, at this scale, even the very little amount of thermal expansion can cause severe problems with prints.  
Hope this helps.

tpipits, 2020-05-19:
Thank you for your response...
I have found that i have problem with x twist (because the skrews of z and z1 endstop was loose and lost position during move so everytime that it home x axis was unlevel). I print new z axis carriages, new skrews and nuts at endstop and new x carriage and my problem solved...

closed 2020-05-19 by tpipits

Selfmade-RuLeZ, 2021-02-16:
have a nearly same issue with my i3 mega S with x-axis twist. Can you provide some stl files for your modification?