Klipper3d/klipper #2872: Community Ettiquette

Hairyloon, 2020-05-14:
I'm fairly new to GitHub, though quite familiar with internet discussions in general.  
I think Klipper is an excellent concept that I'd like to see developed and improved upon, but I don't know what I can do to help and there doesn't seem to be anywhere here for making suggestions: I don't consider suggestions to be "Issues" and they're certainly not "Pull requests".
Where should they go?

theopensourcerer, 2020-05-14:
There a fairly busy channel on IRC. Info here:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

JohnEdwa, 2020-05-16:
IRC is certainly not the proper place, as it's a real-time chat system. It's good for off-topic discussions and troubleshooting but not really anything that actually needs to be properly discussed long term. Even Discord, which really is IRC in disguise would be better as at least you can easily link to and pin previous messages.

But, if/as the goal is mostly to get Kevin's attention, [he would want you to use the mailing list](https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#mailing-list). But as mailing lists have been obsolete for the last 15 years and the secondary goal is to get the community attention as well, they all end up here in Github instead.

Hairyloon, 2020-05-16:
> IRC is certainly not the proper place, as it's a real-time chat system. It's good for off-topic discussions and troubleshooting but not really anything that actually needs to be properly discussed long term.

That was my thinking on that also.

> But, if/as the goal is mostly to get Kevin's attention, [he would want you to use the mailing list]

I imagine that Kevin is a busy chap: I wouldn't necessarily want to get his attention and distract him from something more important.

>(https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#mailing-list). But as mailing lists have been obsolete for the last 15 years and the secondary goal is to get the community attention as well, they all end up here in Github instead.

I wouldn't say they're obsolete: they have their uses, but they are not a good tool for this purpose.

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the conversation has concluded.  FWIW, the only recommendations I, personally, have are in the docs/Contact.md file.

-Kevin

closed 2020-06-14 by KevinOConnor