Klipper3d/klipper #2873: Resume not moving to correct Z position

markwinger, 2020-05-14:
I have setup filament out.  The runout calls a macro called MY_PRINT_PAUSE. In octoprint I setup gcode macros to call this arfer pause.

I then have a macro called MY_PRINT_RESUME the is called from the octoprint resume_filament menu item and the macro is called by octoprint gcode macro before resume.

This seems to work most of the time, but some times when I resume, the z-position is to high.  It looks like the height it was parked during the pause.  When this happens it will print a little while and suddenly is will move the correct level.

I'm not sure what I am doing wrong.  Any suggestions?

klippy.log and printer.cfg(printer.txt) are included.  Printer is an ender 3 with an skr mini e3 v1.2 main board.
[printer.txt](https://github.com/KevinOConnor/klipper/files/4631210/printer.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4631174/klippy.log)

Arksine, 2020-05-14:
First I have a couple of general comments about pause_resume with Octoprint:
1) Make sure you are running the latest version of Octoprint.  It should fix an issue in prior versions where at times it would ignore acknowledgements which lead to issues with PAUSE/RESUME
2) If you are using Octoprint's UI to pause and resume prints, make sure that it isn't sending commands that interfere with Klipper's PAUSE/RESUME functionality.  In the serial settings there is section for "pausing commands".  Make sure that is empty.   Octoprint's "after pause" should contain only your pause macro and "before resume" should contain your resume macro.

With all that said, looking at your log it appears that you have not defined the "MY_PRINT_PAUSE" or "MY_PRINT_RESUME" in printer.cfg, but instead defined "MY_PAUSE_PRINT" and "MY_RESUME_PRINT".  This is what is in your log:
```
Unknown command:"MY_PRINT_PAUSE"
...
Print is not paused, resume aborted
```
Your Octoprint "after pause" script is calling the wrong command, thus Klipper is not internally saving the paused state.  When RESUME is called Klipper knows that PAUSE hasn't been, therefore it can't restore state.    The result is that you are relying on Ocotprint's pause behavior through the UI.  It will drop back to the appropriate Z location on the next layer change.

This also suggests that you are parking the tool in Octoprint's script, and I highly recommend against that.  As I mentioned above, it is ill advised to attempt to mix Octoprint's pause functionality with Klipper's.

markwinger, 2020-05-15:
Octoprint is 1.4.0, according to software update there are no updates available.

In octoprints gcode scripts I have MY_PAUSE_PRINT and MY_RESUME_PRINT and nothing else.  The MY_PRINT_PAUSE had incorrectly been added earlier and I saw the error "Unknown command:"MY_PRINT_PAUSE" and I corrected it.  Notice that was with a previous print.

The last print in the (which had the problem I described) occurred with a filament runout:
Filament Sensor filament_sensor: runout event detected, Time 24508.32
action:paused
(There's more than one so I'm not sure which one faile, but it was near the end)
The log does not seem to show the macro name that was run but in the printer config, runout_gcode: has only one command, MY_PAUSE_PRINT.
Shortly after the runout_gcode is detected I see idle_timeout changed to 20000 which is done in macro MY_TEMPS which is the second line of MY_SUSPEND_PRINT.

Arksine, 2020-05-15:
As I mentioned above, your "MY_PAUSE_PRINT" macro is parking the tool before executing the PAUSE gcode.  This is likely your issue, you need to PAUSE first so it can save the current gcode state before moving the tool.

markwinger, 2020-05-15:
Oh, I get it.  For some reason I was thinking that the last thing I need to do was the pause.  But I need to do every thing before the pause.  I was also getting over extrusion errors sometime too.  Making this change appears to have corrected both.

Thanks.

KevinOConnor, 2020-05-24:
I'm going to close this as it appears a solution was found for the original issue.

-Kevin

closed 2020-05-24 by KevinOConnor