Klipper3d/klipper #2874: Manual and automatic switching of extruders causes an error. Reported error: Move exceeds maximum extrusion (126.066mm ^ 2 vs 0.640mm ^ 2)

Goodfeat, 2020-05-15:
Hello

When setting up the printer for two extruders, I encountered an error. I read topics on this error from other discussion threads, but could not find a solution to my problem.

the bottom line is that I don’t have retracts. I'm just in test mode trying to switch from the first extruder to the second. Moreover, the more time the first extruder takes to print, the more the maximum extrusion error increases.

I use Сura 4.5 (I tried with Repeater and Marlin mode). but the problem remains. I tried to put the command T1
 directly in the g-code in line 360 ​​(I attach the file).
the clipper stubbornly refuses to switch extruders correctly and cancels printing.

judging by what I read, the fix max_extrude_cross_section will not help, because the error is only growing. Adding line G1 F15000 after switching the extruder also does not help

As I understand it, when the extruder switches, it does not actually transfer the position from the first to the second extruder. And since the second extruder is in the zero position, an error is obtained with a difference in the amount of extruded plastic

if I think correctly, how in the clipper to transfer the position from the first to the second extruder? if this is a bug, then how to get around it?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4632137/klippy.log)
[printer.zip](https://github.com/KevinOConnor/klipper/files/4632142/printer.zip)

[bob.stl_T1_line_360.zip](https://github.com/KevinOConnor/klipper/files/4632144/bob.stl_T1_line_360.zip)

dianlight, 2020-05-17:
my2cents:

You can try with section set a plausible **max_extrude_only_distance** in [extruder].

L.

Goodfeat, 2020-05-17:
I seem to have a different solution, it seems to work well, but I'm not sure if it is reliable

```
[gcode_macro T1]
axes: xyz
gcode:
#   M301
#   SET_SERVO SERVO=extruder_servo angle=100    # Lift secondary extruder
	M400
	{% if printer.toolhead.homed_axes == "xyz" %}	
		SET_GCODE_OFFSET X=-9.9 Z={printer.save_variables.zoffsetbed|float + 5} MOVE=1 MOVE_SPEED=100  # Adjust z-height 
		SAVE_GCODE_STATE NAME=EXTR1
		ACTIVATE_EXTRUDER EXTRUDER=extruder1
		RESTORE_GCODE_STATE NAME=EXTR1
	{% endif %}
```

drains SAVE_GCODE_STATE NAME = EXTR0 and RESTORE_GCODE_STATE NAME = EXTR0 simply resume their position after switching.

However, I need to transfer the position of the extruder only, but due to the specifics of the command, the lines transmit the position in all coordinates. Nothing can be done with this.

Not sure if this is the right decision. I have already encountered this on firmware repeater. I needed to fix the firmware code so that the switch was correct. In this firmware I can’t fix anything. Don't know python programming language

KevinOConnor, 2020-05-24:
On a ACTIVATE_EXTRUDER command, Klipper currently resets the "gcode base position for the extruder".  That is, it effectively runs `G90 E0`.  Unfortunately, things like this aren't well defined in g-code.  I'm surprised Cura doesn't explicitly send its own `G90 E0` command.

If you want some other extruder position, it should be possible to do that within the T0 macro (see Command_Templates.md for the details).  For example, something like `G90 E{printer.gcode.gcode_position.e}`.

-Kevin

Goodfeat, 2020-05-25:
> Например, что-то вроде `G90 E{printer.gcode.gcode_position.e}`.

Ok i will try this as an alternative

closed 2020-06-14 by KevinOConnor