Klipper3d/klipper #2878: I installed klipper + dwc2 can't start webserver

gustavya, 2020-05-15:
Hello everyone, I installed klipper+dwc2 and finally flashed a powerspec pro(same as FFCP)

I have the following error in the log file, I guess it's why I can't see the webserver.

Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 118, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1684, in load_config
    return web_dwc2(config)
The RPI seems connected and the board flashed fine, since I can see movement in the lcd, like a wave of lines flowing.

Any advice?
TIA

Gus

gustavya, 2020-05-15:
here is the full log

login as: pi
pi@192.168.1.31's password:
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri May 15 15:24:53 2020 from 192.168.1.63

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated)
   Active: active (running) since Fri 2020-05-15 15:25:05 BST; 1min 18s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 328 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCES
    Tasks: 2 (limit: 2200)
   Memory: 26.2M
   CGroup: /system.slice/klipper.service
           └─372 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.p

May 15 15:25:04 raspberrypi systemd[1]: Starting LSB: Klipper daemon...
May 15 15:25:05 raspberrypi klipper[328]: Starting klipper: klipper.
May 15 15:25:05 raspberrypi systemd[1]: Started LSB: Klipper daemon.
 ESC
pi@raspberrypi:~ $ nano klippy.log
pi@raspberrypi:~ $ cd /tmp
pi@raspberrypi:/tmp $ cd klippy
-bash: cd: klippy: No such file or directory
pi@raspberrypi:/tmp $ dir
klippy.log
printer
systemd-private-4cf61652211748c6b50e74736f27ea19-systemd-timesyncd.service-CLB2P5
pi@raspberrypi:/tmp $ nano klippy.log
pi@raspberrypi:/tmp $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated)
   Active: active (running) since Fri 2020-05-15 15:25:05 BST; 15min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 328 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCES
    Tasks: 2 (limit: 2200)
   Memory: 26.2M
   CGroup: /system.slice/klipper.service
           └─372 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.p

May 15 15:25:04 raspberrypi systemd[1]: Starting LSB: Klipper daemon...
May 15 15:25:05 raspberrypi klipper[328]: Starting klipper: klipper.
May 15 15:25:05 raspberrypi systemd[1]: Started LSB: Klipper daemon.

pi@raspberrypi:/tmp $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated)
   Active: active (running) since Fri 2020-05-15 15:25:05 BST; 19min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 328 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCES
    Tasks: 2 (limit: 2200)
   Memory: 26.2M
   CGroup: /system.slice/klipper.service
           └─372 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.p

May 15 15:25:04 raspberrypi systemd[1]: Starting LSB: Klipper daemon...
May 15 15:25:05 raspberrypi klipper[328]: Starting klipper: klipper.
May 15 15:25:05 raspberrypi systemd[1]: Started LSB: Klipper daemon.

pi@raspberrypi:/tmp $ nano klippy.log
  GNU nano 3.2                       klippy.log

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 118, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1684, in load_config
    return web_dwc2(config)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 49, in __init__
    self.stepper_enable = self.printer.try_load_module(config, "stepper_enable")
AttributeError: Printer instance has no attribute 'try_load_module'

gustavya, 2020-05-15:
just in case here is the full log 

Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-526-gf0c394de-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Start printer at Fri May 15 15:25:07 2020 (1589552707.6 11.7)
===== Config file =====
[stepper_x]
step_pin = PF1
dir_pin = !PF0
enable_pin = !PF2
step_distance = .01038689071
endstop_pin = ^!PL1
position_endstop = 152
position_max = 153
position_min = -120
homing_speed = 50

[stepper_y]
step_pin = PF5
dir_pin = !PF4
enable_pin = !PF6
step_distance = .01038689071
endstop_pin = ^!PL3
position_endstop = 77
position_max = 78
position_min = -84
homing_speed = 50

[stepper_z]
step_pin = PK1
dir_pin = !PK0
enable_pin = !PK2
step_distance = .0025
endstop_pin = !PL6
position_endstop = 0
position_max = 230
position_min = 0

[extruder]
step_pin = PA3
dir_pin = !PA2
enable_pin = !PA4
step_distance = .01038689071
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PH3
sensor_type = MAX6675
sensor_pin = PE3
spi_software_miso_pin = PE5
spi_software_sclk_pin = PE2
spi_software_mosi_pin = PA1
control = pid
pid_kp = 26.414
pid_ki = 1.115
pid_kd = 156.5054
min_temp = 0
max_temp = 260
pressure_advance = 0.07
max_extrude_only_distance = 100

[heater_fan extruder_fan]
pin = PH4

[fan]
pin = PL5

[heater_bed]
heater_pin = PL4
sensor_type = ATC Semitec 104GT-2
sensor_pin = PK7
control = pid
pid_kp = 70.037
pid_ki = 1.710
pid_kd = 717.000
min_temp = 0
max_temp = 130

[mcu]
baud = 115200
serial = /dev/serial/by-id/usb-MakerBot_Industries_The_Replicator_8553130353135151C0D1-if00
restart_method = command

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[mcp4018 x_axis_pot]
scl_pin = PJ5
sda_pin = PF3
wiper = 0.50
scale = 0.773

[mcp4018 y_axis_pot]
scl_pin = PJ5
sda_pin = PF7
wiper = 0.50
scale = 0.773

[mcp4018 z_axis_pot]
scl_pin = PJ5
sda_pin = PK3
wiper = 0.50
scale = 0.773

[mcp4018 a_axis_pot]
scl_pin = PJ5
sda_pin = PA5
wiper = 0.50
scale = 0.773

[mcp4018 b_axis_pot]
scl_pin = PJ5
sda_pin = PJ6
wiper = 0.50
scale = 0.773

[virtual_sdcard]
path = /home/pi/sdcard

[web_dwc2]
printer_name = woot
listen_adress = 192.168.1.31
listen_port = 4750
web_path = dwc2/web
=======================
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 136, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 129, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 118, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 1684, in load_config
    return web_dwc2(config)
  File "/home/pi/klipper/klippy/extras/web_dwc2.py", line 49, in __init__
    self.stepper_enable = self.printer.try_load_module(config, "stepper_enable")
AttributeError: Printer instance has no attribute 'try_load_module'

jakep82, 2020-05-15:
You need to report this to Stephan at his DWC2 repo.  DWC2 is not supported by Klipper.

https://github.com/Stephan3/dwc2-for-klipper/issues

icstra, 2020-05-15:
Get the last release. Files in klippy/extras have been updated

icstra, 2020-05-15:
Modify line 49 in dwc2-for-Klipper.py. Replace try to load module with load_object

RomanLyavin, 2020-05-17:
> Modify line 49 in web_dwc2.py. Replace try_to_load module with load_object

senks!

shaiss, 2020-05-20:
> > Modify line 49 in web_dwc2.py. Replace try_to_load module with load_object
> 
> senks!

confirmed that works!   ty!

closed 2020-05-22 by klipper-gitissuebot