Klipper3d/klipper #2879: Make segmentation fault

KenDodman, 2020-05-15:
Fresh install of octopi

Here is my command history 
    1 git clone https://github.com/KevinOConnor/klipper
    2  ./klipper/scripts/install-octopi.sh
    3  cd ~/klipper/
    4  make menuconfig
    5  make

pi@octopi:~/klipper $ make
  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/spi_software.o
  Compiling out/src/neopixel.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/serial.o
  Compiling out/src/generic/serial_irq.o
make: *** [Makefile:86: out/src/sched.o.ctr] Segmentation fault

KenDodman, 2020-05-15:
resolved by starting with a fresh sd, and latest octopi image, took 2 trys, didnt get any errors on any installs either time

      sudo apt update
      sudo apt full-upgrade
      git clone https://github.com/KevinOConnor/klipper
      ./klipper/scripts/install-octopi.sh
      cd ~/klipper/
      make menuconfig
      make
      ls /dev/serial/by-id/*
     sudo service klipper stop
     make flash FLASH_DEVICE=/dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Mega_2560_95333303131351501252-if00
     sudo service klipper start

KevinOConnor, 2020-05-24:
FYI, these types of errors are typically due to failing sd card or a bad OS install.  Looks like you've found a solution though, so I'm going to close this.

-Kevin

closed 2020-05-24 by KevinOConnor

o-nix, 2021-10-01:
> these types of errors are typically due to failing sd card

This!
Now I finally understand why I had been getting random `Unzip errors` + the segmentation fault issue above.
New SD card -> working well.
Many thanks to all!