Klipper3d/klipper #2880: The dreaded "MCU 'mcu' shutdown: Stepper too far in past" error

pietrobaricco, 2020-05-16:
I've had several failed prints recently because of this issue. I have a dual extruder printer, one of the extruders (the one giving the problem) is a fairly long bowden which requires a pressure advance of around 1. It also has a 0.9 degrees stepper, and it is driven by a mks s-base at 32 microstepping.
I have a mixture of TMC2130 drivers in standalone mode and embedded DRVs, so can't compile with the delay set to 0.
I have now replaced the motor with another one at 1.8 degrees, printing the same problematic gcode: will report back the results once it finishes, or fails.
I have observed it happening mostly during slow moves on rounded curves, most often printing skirts, which seem to have a huge number of very small segments (I use S3D).
I am wondering, what would happen if I comment the shutdown call in stepper.c and just keep on printing, maybe outputting something on the log? 
A slight skip on a stupid tiny extruder move may be completely unnoticeable, and sure it would be better than a failed print.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4638133/klippy.log)

KevinOConnor, 2020-05-17:
It sounds like you found a solution.  However, if you're looking for some recommendations: If you can, I'd reduce the micro-stepping on your extruder - nominally stepping 333nanometers per step adds extra processing time without a quality improvement.  I'd also recommend updating to the latest code as there was an issue (commit 7843c23b) that could be making things worse for you.  

-Kevin

pietrobaricco, 2020-05-17:
Unfortunately I can't reduce it further as the extruder with 0.00333 step_distance is a zesty nimble, which has a ridiculous 30:1 reduction. 
It's been working fine though, probably because it requires sane (small) amounts of  pressure advance.
I could set the board to 16 microsteps instead of 32, but I'd prefer to keep it at 32 as it's noticeably quieter and smoother set up that way.
My question / issue is more about why Klipper shuts down the printer when this error occurs instead of trying to somehow recover and save the print?

KevinOConnor, 2020-05-24:
FWIW, I want my printer to tell me when there is a problem.  That way I can fix the underlying issue and obtain quality prints.  I definitely do not want my printer to continue and silently produce bad quality prints.

In the above situation, the micro-controller was sent a series of commands that it could not achieve - its schedule got hopelessly out of sync with the requested schedule.  It therefore informed the host of the error.

-Kevin

closed 2020-05-25 by pietrobaricco