Klipper3d/klipper #2883: Request for the function of reading octoprint files from the display menu

Goodfeat, 2020-05-17:
Hello.
Is there any way to add print start files via the display menu?

Octoprint has a number of uploaded files, is it possible to get a list of files uploaded in octoprint through the display menu?

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/4639488/klippy.log) Emptyfile

KevinOConnor, 2020-05-24:
It's possible if one enables virtual_sdcard support.  There's also #1392 .  Separately, the menu is in flux (and has been for over a year), so it isn't currently a target for new features.

-Kevin

Goodfeat, 2020-05-25:
It seems to work, but not quite as I wanted.
The list in octoprint is sorted by date added, while the display shows everything in alphabetical order. It seems that setting the sorting function would not hurt

P.S read several topics on the sorting problem, in particular topic #1387.

In the mentioned topic, you seem to have not agreed with the innovation of sorting by name, date and file size.

In this case, the sorting of the octoprint does not match the sorting in the display.

In the script there is a sort only by file name. I tried adding a piece from the script from @trevjonez. This code is for sorting files by name and it seems to work just fine. It may be worth adding the function of sorting files by date and size in the official version of the Klipper?

```
 sort_choices = {'name': self._sort_by_name,
                        'date': self._sort_by_date,
                        'size': self._sort_by_size}
        self.sort = config.getchoice('sort_by', sort_choices, 'name')
        self.sort_reverse = config.getboolean('sort_reverse', False)
```
```
    def get_file_list(self):
        dname = self.sdcard_dirname
        try:
            raw_dir_list = os.listdir(self.sdcard_dirname)
            filtered = filter(lambda fname: not fname.startswith('.')
                                            and fname.endswith('.gcode'),
                              raw_dir_list)
            sorted_files = sorted(filtered,
                               key=lambda fname: self.sort(dname, fname),
                               reverse=self.sort_reverse)
            return [(fname, os.path.getsize(os.path.join(dname, fname)))
                    for fname in sorted_files]
        except:
            logging.exception("virtual_sdcard get_file_list")
            raise self.gcode.error("Unable to get file list")
    def _sort_by_name(self, dname, fname):
        return fname
    def _sort_by_date(self, dname, fname):
        #negate mtime so that default is descending
        return os.path.getmtime(os.path.join(dname, fname)) * -1
    def _sort_by_size(self, dname, fname):
        #negate size so that default is descending
        return os.path.getsize(os.path.join(dname, fname)) * -1
```

lbibass, 2020-05-27:
I agree. not being able to sort by date uploaded makes the virtualSD essentially useless, at least for me.

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the conversation has concluded.  FWIW, I understand that the menu code could use some additional features, but some base work needs to be done to the menu before any additional features will be considered.

-Kevin

closed 2020-06-14 by KevinOConnor