Klipper3d/klipper #2884: Move out of range X

gustavya, 2020-05-17:
I'm trying to configure Klipper since 2 days(powerspec pro, FFPC clone) and it seems I'm very bad at it, becouse it's been 3 days.
Now I can't figure out how to solve this, I have this error becouse in the config file I have 

[stepper_x]
step_pin: PF1
dir_pin: !PF0
enable_pin: !PF2
step_distance: .01038689071
endstop_pin: ^!PL1
position_endstop: 152
position_max: 153 **_(if I put 157 it gives me no error, but the printer smashes the extruder to the right)_**
position_min: -120
homing_speed: 50

I can't find where to change max travel, I looked into simplify3d settings(changed everything here), octoprint settings(didn't find anything to change)

Hope anybody can help, this instalation is driving me crazy, girlfriend is mad

`Starting Klippy...
Args: ['/home/pi/klipper-flashforge-creator-pro/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.8.0-360-gb3723065-dirty'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.16 (default, Apr  6 2019, 01:42:57) \n[GCC 8.2.0]'
Start printer at Sun May 17 03:59:37 2020 (1589684377.8 10.1)
===== Config file =====
[stepper_x]
step_pin = PF1
dir_pin = !PF0
enable_pin = !PF2
step_distance = .01038689071
endstop_pin = ^!PL1
position_endstop = 152
position_max = 153
position_min = -120
homing_speed = 50

[stepper_y]
step_pin = PF5
dir_pin = !PF4
enable_pin = !PF6
step_distance = .01038689071
endstop_pin = ^!PL3
position_endstop = 77
position_max = 78
position_min = -84
homing_speed = 50

[stepper_z]
step_pin = PK1
dir_pin = !PK0
enable_pin = !PK2
step_distance = .0025
endstop_pin = !PL6
position_endstop = 0
position_max = 230
position_min = 0

[extruder]
step_pin = PA3
dir_pin = !PA2
enable_pin = !PA4
step_distance = .01038689071
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PH3
sensor_type = MAX6675
sensor_pin = PE3
spi_software_miso_pin = PE5
spi_software_sclk_pin = PE2
spi_software_mosi_pin = PA1
[... 90 lines cut ...]
Unknown command:"G161"
Unknown command:"G161"
Unknown command:"M132"
Unknown command:"M126"
Unknown command:"M116"
Unknown command:"G130"
Move out of range: 157.000 -63.000 35.000 [0.000]
Unknown command:"M108"
Stats 83.7: gcodein=493 mcu: mcu_awake=0.017 mcu_task_avg=0.000120 mcu_task_stddev=0.000105 bytes_write=4648 bytes_read=12459 bytes_retransmit=9 bytes_invalid=0 send_seq=300 receive_seq=300 retransmit_seq=2 srtt=0.005 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=70 temp=32.1 pwm=1.000  print_time=138.169 buffer_time=6.904 print_stall=0 extruder: target=200 temp=36.0 pwm=1.000
Heater heater_bed approaching new target of 70.000
Heater extruder approaching new target of 200.000
Stats 84.7: gcodein=493 mcu: mcu_awake=0.024 mcu_task_avg=0.000141 mcu_task_stddev=0.000131 bytes_write=4669 bytes_read=12582 bytes_retransmit=9 bytes_invalid=0 send_seq=302 receive_seq=302 retransmit_seq=2 srtt=0.004 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000137 heater_bed: target=70 temp=35.9 pwm=1.000  print_time=138.169 buffer_time=5.903 print_stall=0 extruder: target=200 temp=35.8 pwm=1.000
Stats 85.7: gcodein=493 mcu: mcu_awake=0.024 mcu_task_avg=0.000141 mcu_task_stddev=0.000131 bytes_write=4675 bytes_read=12685 bytes_retransmit=9 bytes_invalid=0 send_seq=303 receive_seq=303 retransmit_seq=2 srtt=0.004 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=70 temp=36.7 pwm=1.000  print_time=138.169 buffer_time=4.903 print_stall=0 extruder: target=200 temp=35.5 pwm=1.000
Stats 86.7: gcodein=493 mcu: mcu_awake=0.024 mcu_task_avg=0.000141 mcu_task_stddev=0.000131 bytes_write=4681 bytes_read=12817 bytes_retransmit=9 bytes_invalid=0 send_seq=304 receive_seq=304 retransmit_seq=2 srtt=0.004 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=70 temp=37.9 pwm=1.000  print_time=138.169 buffer_time=3.902 print_stall=0 extruder: target=200 temp=35.5 pwm=1.000
Stats 87.7: gcodein=493 mcu: mcu_awake=0.024 mcu_task_avg=0.000141 mcu_task_stddev=0.000131 bytes_write=4719 bytes_read=12930 bytes_retransmit=9 bytes_invalid=0 send_seq=307 receive_seq=307 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=70 temp=34.0 pwm=1.000  print_time=138.169 buffer_time=2.901 print_stall=0 extruder: target=200 temp=35.8 pwm=1.000
Stats 88.7: gcodein=493 mcu: mcu_awake=0.024 mcu_task_avg=0.000141 mcu_task_stddev=0.000131 bytes_write=4725 bytes_read=13033 bytes_retransmit=9 bytes_invalid=0 send_seq=308 receive_seq=308 retransmit_seq=2 srtt=0.004 rttvar=0.002 rto=0.025 ready_bytes=24 stalled_bytes=0 freq=16000139 heater_bed: target=70 temp=36.5 pwm=1.000  print_time=138.269 buffer_time=2.000 print_stall=0 extruder: target=200 temp=36.0 pwm=1.000
Stats 89.7: gcodein=540 mcu: mcu_awake=0.011 mcu_task_avg=0.000116 mcu_task_stddev=0.000115 bytes_write=4781 bytes_read=13189 bytes_retransmit=9 bytes_invalid=0 send_seq=311 receive_seq=311 retransmit_seq=2 srtt=0.004 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000139 heater_bed: target=0 temp=36.7 pwm=0.000  print_time=138.269 buffer_time=0.999 print_stall=0 extruder: target=0 temp=36.5 pwm=0.000
Stats 90.7: gcodein=540 mcu: mcu_awake=0.011 mcu_task_avg=0.000116 mcu_task_stddev=0.000115 bytes_write=4800 bytes_read=13297 bytes_retransmit=9 bytes_invalid=0 send_seq=313 receive_seq=313 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000139 heater_bed: target=0 temp=38.2 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=37.0 pwm=0.000
Stats 91.7: gcodein=540 mcu: mcu_awake=0.011 mcu_task_avg=0.000116 mcu_task_stddev=0.000115 bytes_write=4806 bytes_read=13400 bytes_retransmit=9 bytes_invalid=0 send_seq=314 receive_seq=314 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000139 heater_bed: target=0 temp=31.5 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=37.2 pwm=0.000
Stats 92.7: gcodein=540 mcu: mcu_awake=0.011 mcu_task_avg=0.000116 mcu_task_stddev=0.000115 bytes_write=4812 bytes_read=13532 bytes_retransmit=9 bytes_invalid=0 send_seq=315 receive_seq=315 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=0 temp=31.6 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=38.8 pwm=0.000
Stats 93.8: gcodein=540 mcu: mcu_awake=0.011 mcu_task_avg=0.000116 mcu_task_stddev=0.000115 bytes_write=4818 bytes_read=13635 bytes_retransmit=9 bytes_invalid=0 send_seq=316 receive_seq=316 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=0 temp=31.1 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=39.2 pwm=0.000
Stats 94.8: gcodein=545 mcu: mcu_awake=0.009 mcu_task_avg=0.000102 mcu_task_stddev=0.000103 bytes_write=4824 bytes_read=13752 bytes_retransmit=9 bytes_invalid=0 send_seq=317 receive_seq=317 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=0 temp=31.2 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=39.8 pwm=0.000
Stats 95.8: gcodein=545 mcu: mcu_awake=0.009 mcu_task_avg=0.000102 mcu_task_stddev=0.000103 bytes_write=4830 bytes_read=13884 bytes_retransmit=9 bytes_invalid=0 send_seq=318 receive_seq=318 retransmit_seq=2 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=0 temp=30.9 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=40.8 pwm=0.000
Stats 96.8: gcodein=545 mcu: mcu_awake=0.009 mcu_task_avg=0.000102 mcu_task_stddev=0.000103 bytes_write=4836 bytes_read=13987 bytes_retransmit=9 bytes_invalid=0 send_seq=319 receive_seq=319 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000138 heater_bed: target=0 temp=31.1 pwm=0.000  print_time=138.269 buffer_time=0.000 print_stall=0 extruder: target=0 temp=41.5 pwm=0.000
`

BlackStump, 2020-05-17:
[Read the Example Config](https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L23)
[homing_positive_dir](https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L65)

BlackStump, 2020-05-17:
[Config_checks](https://www.klipper3d.org/Config_checks.html)

closed 2020-05-24 by klipper-gitissuebot