Klipper3d/klipper #2885: [FEATURE REQUEST]  Filament switch sensor add jam functionality

test3210-d, 2020-05-17:
I think correctly add encoder based jam sensor functionality internal to  switch sensor module rather than create a separate module

 operation principle
any change on button pin increase counter, check_dist  -  parameter from config, if no changes from encoder on check distance filament jammed

```
    def jam_button_handler(self, eventtime, state):
        self.counter=self.counter + 1

    def extrude_factor_update_event(self, eventtime):
        # Update extrude factor
        pos = self.toolhead.get_position()
        epos = pos[3]
        if epos == 0:
            self.old_dist=self.check_dist
        if self.old_dist<epos:
            self.old_dist=epos+self.check_dist
            self.filament_present = (self.counter > 0)
            self.counter=0
            return eventtime + 1
```

Arksine, 2020-05-19:
This may work well enough for jam detection if "check_dist" is a large enough value.  The issue you run into here is that the toolhead position is not the current e-position, but some position in the future due to lookhead.  The encoder on the "jam" sensor is measuring realtime filament movement and that is being compared to an e-position that the stepper has not yet reached.  This could easily generate false positives when the lookahead is buffer is filling up.

The downside is that it would take some time for an actual jam to be detected and initiate the PAUSE.  If the goal is to simply stop printing when a jam is detected this can work.  If the goal is to give users an opportunity to clear the jam and save the print this may not work out well.

test3210-d, 2020-05-19:
as i know most jam sensors have check distance 4mm or more. Sensor generate 1 low signal on 1 or 2 mm.  Klipper generate event on evry change on the pin, it is 2 count per 2 mm if sensor step 2 mm. 
I think accurately is enough.

Arksine, 2020-05-19:
Right, but the issue is you are comparing that 4mm of realtime movement to a buffered position.

Let's say you set a check distance to 5mm.  Whenever the toolhead is incremented +5mm your timer will check the encoder's counter to make sure that there is no jam.

Now lets move to Klipper.  Lets say it gets a gcode requesting a 7mm extrusion move.  Assuming the toolhead isn't stalled waiting for current moves to process, that 7mm move gets queued and the toolhead's e position is adjusted the full 7mm immediately.  When your timer does the check it will see that the toolhead's position is +7mm, but the encoder didn't trigger because the extruder has not yet moved that 7mm.  You have a false positive because the toolhead's position is not the realtime position.

test3210-d, 2020-05-19:
That is, there is no easy solution with a high accuracy sensor?

Arksine, 2020-05-20:
For an accurate comparison you need to get the realtime e-axis position.  This is possible by querying the mcu's e-stepper position, but that is not likely something you want to do during a print.  For the PAT9125 I pushed the mcu e-position along with the readout of the sensor to the host, which works but is not ideal (this is why I havent created a PR for it).

Given that the host schedules step times it is possible for the host to track the realtime stepper position without querying the MCU, however such functionality would require implementation.

kakou-fr, 2020-05-29:
i am also working on it, i do a patch for the firmware to report the current position of the extruder and the filament encoder (like for temperature sensor) every 0.3s. 

My sensor use an arduino nano that report the length mesured by the rotary sensor, and it's connected via SPI to the board (i will maybe switch to I2C).

For now, i am able to detect jamming in less that 1-2 seconds.

KevinOConnor, 2020-06-25:
I think we can mark this as inactive until a patch is ready.

-Kevin

closed 2020-06-25 by KevinOConnor