Klipper3d/klipper #2886: proposal: programmatically interact with Menu

rpanfili, 2020-05-17:
I would love to being able to interact with the menu component via the `printer.menu` variable.
This could be useful to develop some kind of menu funnel like an _interactive wizards_, etc.

To being able to try this proof of concept I exposed three new actions on the `MenuManager` component via the `get_status` method:
```python
	     'is16xx': (self.cols == 16),

            'action_back': self.back,
            'action_exit': self.exit,
            'action_load_root': self.load_root

       }}
```
[not sure that `load_root` is the best way to change loaded tree, not tested yet]

Is this a bad idea or too hackish? Are there any side effect?

rpanfili, 2020-05-18:
[since it's just a feature, not a bug I'm adding a dummy log to avoid a premature closure of this thread]
[klippy.log](https://github.com/KevinOConnor/klipper/files/4642793/klippy.log)

KevinOConnor, 2020-05-24:
FYI, the menu is in flux (and has been for over a year).  There's a goal to update it to the Jinja2 syntax.  Unfortunately, until we can get that complete, it's not a target for new features.

-Kevin

closed 2020-05-26 by rpanfili