Klipper3d/klipper #2889: MCU 'mcu' shutdown: Timer too close

Hairyloon, 2020-05-18:
This happened several hours into a print. It was quite convenient as it happened because the print had failed so it saved me some mess.
I do not believe the failure is related to the error, nor vice versa: I suspect a slipped belt, but I'd like to know what the error means and how to fix it.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4643577/klippy.log)

The log file is almost 20MB and most of it says much the same, so I've chopped a lot of it out.

Hairyloon, 2020-05-20:
Anyone?

KevinOConnor, 2020-05-24:
Alas, there's nothing we can do without the log from the event itself.  If it's large you can zip the log prior to uploading.  Also, your sample log doesn't show the software versions - note that we also need that info as well in order to perform any meaningful investigation.

As to the cause of the typical cause of the error - that should have been described along with the error itself:
```
This is generally indicative of an intermittent
communication failure between micro-controller and host.
```

-Kevin

Hairyloon, 2020-05-24:
> Alas, there's nothing we can do without the log from the event itself. If it's large you can zip the log prior to uploading. Also, your sample log doesn't show the software versions - note that we also need that info as well in order to perform any meaningful investigation.
> 
> 

Ah, sorry I'd assumed the event had happened just before the error reporting it. Perhaps there was an earlier report that I'd missed.
I can find out the software versions, but I think I don't have that log any more.

closed 2020-05-24 by Hairyloon