Klipper3d/klipper #2890: Are there any pin constraints for single pin UART for TMC2208

james-fry, 2020-05-18:
This is not an issue, but I cannot find the info I need in docs, code comments (not the actual code) or searching issues.
(so maybe a doc defect? ;) )
I looked here:

My question: Do the pins for single pin UART for TMC2208 need to be interrupt pins?
I have a GT2560 board, and I would like to use X max and Y max pins for UART to TMC2208 X and Y steppers. 
These are NOT interrrupt pins.m (PA2 AD2/ar24 and PA4 AD4/ar26)
(also can see gt2560 schematic here: https://www.geeetech.com/forum/download/file.php?id=4247)

Will this work?
Or do UART pins need to be interrupt enabled pins?
If they DO need to be interrrupt enabled, do they need to be full INT, or is pin change (PCINT) OK?

jakep82, 2020-05-18:
That website is not official documentation.  Any gpio pin should work for UART on a 2208.

james-fry, 2020-05-18:
> That website is not official documentation.
Good to know! I did search github too of course :)

> Any gpio pin should work for UART on a 2208.
Awesome news! I will embark on this quest...

Thanks for the info. I will pop back and report/close issue.