Klipper3d/klipper #2891: Delta Calibrate Issue - Nozzle higher between towers, lower under the tower.

NotExpectedYet, 2020-05-18:
Hello, first time delta user but long time klipper user. Having a wee issue where I can't seem to get klipper to print level at all. 

I've been at this a few days now so really hoping it's just something silly I'm missing as I can't currently get a print off, nor one to do the advanced delta calibration. 

I've done the paper test a fair few times now, and even done my usual checking the level by eye and whenever the delta goes to print it seems to raise the head between the towers causing filament not to stick and then rubs up against the bed. I renamed my config.cfg to config.log as it wouldn't upload otherwise. 

Any help would be appreciated. I've attached both my config and logs for checking.

[klippyDelta.log](https://github.com/KevinOConnor/klipper/files/4644633/klippyDelta.log)
[config.log](https://github.com/KevinOConnor/klipper/files/4644638/config.log)

NotExpectedYet, 2020-05-19:
Just to add to this Ive just tried manually calibrating the towers endstops and got them all to hit perfect level at Z0. 

Now after doing this X0 Y0 Z0 was about 3mm convex compared. So I tried bumping the delta_radius up but that doesn't seem to do anything at all. I went from 140.8 -> 200 and it still sat about 3mm above the center of the bed at Z0. 

Does the delta radius not work the same as it does in marlin?

BlackStump, 2020-05-19:
Are you restarting the firmware after a delta radius change?

Critical factors for a delta calibrate and I would say all firmwares.
1 ~ know the precise hinge to hinge length of your diagonal rods.
2 ~ have a perfectly flat bed perpendicular to the towers if not expect to do some figure fudging.
3 ~ make sure the tower stepper drive mechanism is calibrated ie if GT2 belts correct tension
4 ~ make sure your Z steps are perfectly calibrated before any attempt at delta calibration.

The diagonal rod length is a critical factor if they can be precisely measured then it will be one parameter that can be entered with confidence and not changed.
If your bed is not perfectly flat you will have to account for that when doing your calibration.

do not attempt to use bed mesh until you have calibrated your delta.

NotExpectedYet, 2020-05-20:
Thanks for the reply!

I'm using the OctoKlipper Plugin which does run a restart after each change, but I'll double check that again later on today to double make sure it is actually refreshing. Thought it was strange that delta radius did nothing. 

The bed's perfectly flat and my set square would lead me to believe it's perpendicular but I will double check all you have said and enter those values into the config and try again :)

KevinOConnor, 2020-05-24:
The delta radius is automatically calculated by the DELTA_CALIBRATE command.  So, you don't need to (and don't want to) manually customize that.

It does look like the tool itself isn't finding a great solution for the delta parameters.  Double check that the mechanics are stable (eg, no loose screws) and that the steppers aren't losing steps.  Also, double check that you've got a good starting value for the length of the diagonal rods (the arm lengths).

-Kevin

NotExpectedYet, 2020-05-25:
Thanks for the clarification, I've double checked all the mechanics and I've managed to get a rod length calibration print off which seemed to have helped with the DELTA_CALIBRATE after that words. 

I think this could possibly be me trying to print right at the edge of my bed. If I stay within s3d's set 183.8*183.8 square then it prints just fine. When I travel outside of that I see the issue described above. I'll close this as I believe it's down to my lack of understanding of delta's. Thanks for the help, I'll keep playing and see if I can get it to print outside of that by playing with the slicer settings but for now it's printing and staying level. Will be doing the advanced calibrate shortly.

closed 2020-05-25 by NotExpectedYet