Klipper3d/klipper #2892: CAN bus interface needs txqueusize

bondus, 2020-05-18:
According to docs/stm32f0_CAN.md the CAN interface in linux should be set up as:

auto can0
iface can0 can static
    bitrate 500000

But with some load on the bus there will be too many tx errors due to lost data. The sendserial program tunneling serial over CAN drops data if the tx buffer is full. Adding a txbuffersize option to the interface fixes that problem:

auto can0
iface can0 can static
        bitrate 500000
        txqueuelen 1024

Other than that the CAN port for STM32F042 works perfectly fine, with some small changes it can be made to work on STM32F103 as well.

KevinOConnor, 2020-05-19:
Interesting.  @Delsian - FYI.

-Kevin

Delsian, 2020-05-21:
Thanks, @bondus,  will do this in next commit

bondus, 2020-05-25:
Looks like my fix has bugs. My changes to the network interface definition has no effect.

What actually needs to be run is an: 
ip link set can0 txqueuelen 1024
1024 is probably a bit too much.

KevinOConnor, 2020-06-25:
Is there anything further needed on this github issue?

-Kevin

bondus, 2020-06-26:
It's on my TODO list to update this doc with the changes I made to CAN, I'll get to that point eventually. You can close the issue if you want, but the problem is still there (among many others).
We should probably break out the CAN stuff from the STM32F0/STLINK parts.

bondus, 2020-08-25:
I'll update the docs once my CAN bus boards are done and ready for shipping.

closed 2020-08-25 by bondus