Klipper3d/klipper #2895: Extrude command only works once after each `G92 E0`

aaltman, 2020-05-19:
[single-extrusion-klippy.log](https://github.com/KevinOConnor/klipper/files/4652126/single-extrusion-klippy.log)

This is on an SKR v1.4 Turbo with TMC5160 drivers.  X, Y and Z axis movement work.  The extruder works as well, but only for a single move.  Attempts after the first show:

`passed: G1`

I can trigger this manually but I think it may also be causing ghost prints - I have not been able to get a short job to print yet even though the axis motion works correctly.  What can I try?

jakep82, 2020-05-19:
I believe by default the extruder is in absolute positioning mode (M82).  If you want to issue a G1 command to extrude filament without resetting the position of the extruder (G92 E0), you need to switch to relative mode (M83).

aaltman, 2020-05-19:
I'll try that.  Does that also need to be done in startup gcode when generated from Cura?

aaltman, 2020-05-19:
It works when manually extruding!  Thank you.

The first handful of lines of gcode generated by Cura include an M82.  When I run this, it ghost prints - and I can feel with my hand that the filament is not being moved by the extruder.  Is Cura's gcode incorrect?

`;FLAVOR:Marlin
;TIME:191
;Filament used: 0.0111235m
;Layer height: 0.1
;MINX:188.487
;MINY:188.325
;MINZ:0.1
;MAXX:211.519
;MAXY:211.574
;MAXZ:2.2
;Generated with Cura_SteamEngine 4.2.1
M140 S45
M105
M190 S45
M104 S220
M105
M109 S220
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:22
;LAYER:0
M107
G0 F3600 X191.898 Y191.337 Z0.1
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X192.478 Y190.795 E0.00249
G1 X192.805 Y190.534 E0.0038
G1 X193.284 Y190.171 E0.00568
`

jakep82, 2020-05-20:
I don't know why it's issuing multiple G92 E0 commands in a row.  Otherwise nothing jumps out at me except I wouldn't recommend a first layer height of 0.1.  You may have your nozzle too close to the bed which is easy to do at that layer height, and the result will be a clogged hotend.

I don't use Cura so I won't be able to offer any help with slicer settings.

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the original user is using DWC2.  This issue tracker is exclusively for use with the code from this repo.  The original issue will need to be reproduced with the pristine code, or the issue will need to be reported to the DWC2 developers.

-Kevin

closed 2020-06-14 by KevinOConnor