Klipper3d/klipper #2897: (Voron) Weirdest ADC out of range (no upward or downward spikes from thermistor)

BelgarionNL, 2020-05-19:
I have tried everything. I have replaced the thermistor of the hotend + cabling.

could it be the bed thermistor? 

klippy.log (50mb): https://file.io/F6zF4g69

BelgarionNL, 2020-05-19:
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/4653039/klippy.2.log)

BelgarionNL, 2020-05-19:
both temps seem to be OK:

`Stats 111599.1: gcodein=101203651 mcu: mcu_awake=0.125 mcu_task_avg=0.000313 mcu_task_stddev=0.000286 bytes_write=172620372 bytes_read=21845546 bytes_retransmit=0 bytes_invalid=3 send_seq=2892178 receive_seq=2892178 retransmit_seq=0 srtt=0.005 rttvar=0.002 rto=0.025 ready_bytes=33 stalled_bytes=0 freq=15970103 z: mcu_awake=0.017 mcu_task_avg=0.000144 mcu_task_stddev=0.000168 bytes_write=27211879 bytes_read=7183624 bytes_retransmit=9 bytes_invalid=7 send_seq=600074 receive_seq=600074 retransmit_seq=2 srtt=0.003 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16009996 adj=16039978 heater_bed: target=103 temp=103.4 pwm=0.013 print_time=68892.800 buffer_time=1.932 print_stall=63 extruder: target=243 temp=242.9 pwm=0.532
Stats 111600.1: gcodein=101205043 mcu: mcu_awake=0.125 mcu_task_avg=0.000313 mcu_task_stddev=0.000286 bytes_write=172622502 bytes_read=21845849 bytes_retransmit=0 bytes_invalid=3 send_seq=2892214 receive_seq=2892214 retransmit_seq=0 srtt=0.005 rttvar=0.001 rto=0.025 ready_bytes=49 stalled_bytes=0 freq=15970103 z: mcu_awake=0.017 mcu_task_avg=0.000144 mcu_task_stddev=0.000168 bytes_write=27212687 bytes_read=7183752 bytes_retransmit=9 bytes_invalid=7 send_seq=600089 receive_seq=600089 retransmit_seq=2 srtt=0.005 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16009995 adj=16039969 heater_bed: target=103 temp=103.2 pwm=0.112 print_time=68893.926 buffer_time=2.059 print_stall=63 extruder: target=243 temp=242.3 pwm=0.678
MCU 'mcu' shutdown: ADC out of range`

KevinOConnor, 2020-05-24:
Unfortunately, the attached log is incomplete.  If you want us to take a look at it you will need to attach a complete log from the event.

-Kevin

closed 2020-06-14 by KevinOConnor