Klipper3d/klipper #2899: Internal error on command:"PID_CALIBRATE"

ludobi92, 2020-05-20:
Hello everyone, I try to Calibrate my heatbed and I get the following error, could you help me to solve the issue.

Many thanks in adavance.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4657768/klippy.log)

jakep82, 2020-05-20:
From your log:

```
Heater heater_bed not heating at expected rate
Transition to shutdown state: Heater heater_bed not heating at expected rate
See the 'verify_heater' section in config/example-extras.cfg
for the parameters that control this check.
```

Your bed is heating too slow for the default verify_heater settings.

ludobi92, 2020-05-20:
I have already checked, and I am probably wrong but nothing seems to be abnormal. Idon't know how to fix that

jakep82, 2020-05-20:
Are you sure your bed can actually reach 110C?  Your log indicates it took more than a minute to go from 65.4 to 67.4.  If you believe it can reach 110, then you need to increase the check_gain_time.  By default the bed needs to increase by 2C every 60 seconds or it will fail the check and result in an error.

ludobi92, 2020-05-20:
Many thanks for your helpfull answers, I will try to increase this parameter.

KevinOConnor, 2020-06-14:
I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2020-06-14 by KevinOConnor