Klipper3d/klipper #2900: Recv: // Lost communication with MCU 'mcu'

Hairyloon, 2020-05-20:
As it says in the title. 
For no apparent reason, several hours into a long print...

Any clues?
I don't want a repeat performance if we can avoid it.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4658672/klippy.log)

jakep82, 2020-05-20:
This is generally indicative of a hardware issue.  See the FAQ for places to look.

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors

Hairyloon, 2020-05-21:
Given that I have been having undervoltage warnings and the low probability of finding any more clues (unless it happens again), I'll assume it was that and close this.
Thank you for the prompt response.

closed 2020-05-21 by Hairyloon