Klipper3d/klipper #2903: Huge Z Adjustments when using Z_TILT_ADJUST

danjo-h, 2020-05-21:
I'm trying to get Z_TILT_ADJUST to work on a i3 style printer with independently driven Z motors and a bltouch, mesh leveling works fine, as does using the bltouch as a virtual endstop.

When running z_tilt_adjust, it probes the two points I've set okay, and then calculates a huge z adjust for each stepper. The value seems to sometimes be positive or negative, if its positive it drives the z axis up into the top of the frame.

`Send: z_tilt_adjust
Recv: // probe at 10.000,0.000 is z=0.975000
Recv: // probe at 170.000,0.000 is z=0.967500
Recv: // Making the following Z adjustments:
Recv: // stepper_z = -18353420.733733
Recv: // stepper_z1 = -18353420.749295
Recv: // Retries: 0/10 Probed points range: 0.007500 tolerance: 0.010000`

I've attached the log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4664286/klippy.log)

jakep82, 2020-05-21:
I'm not sure if it's causing your issue, but you should not define position_endstop in your stepper_z section if you're using a bltouch as your endstop.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L552

danjo-h, 2020-05-21:
I've removed it as its better not have things i shouldn't in there. It hasn't made a difference to the strange offset values:

'Send: z_tilt_adjust
Recv: // probe at 10.000,0.000 is z=0.995000
Recv: // probe at 170.000,0.000 is z=1.007500
Recv: // Making the following Z adjustments:
Recv: // stepper_z = 49699183.131812
Recv: // stepper_z1 = 49699183.157750'

KevinOConnor, 2020-05-24:
The z_tilt code was designed to work in both the X and Y dimensions.  So, even though you are only adjusting two motors, you currently need to define 3 probe points and make sure there is diversity in both the X and Y probed point positions.

-Kevin

danjo-h, 2020-05-24:
Thank you!

That makes a lot of sense, it seems like having both the points on Y0 caused the problem. I originally tried with three points but both the front ones were at Y0. Thank you again.

closed 2020-05-24 by danjo-h