Klipper3d/klipper #2906: Adding new kinematics class

Bernouillle, 2020-05-22:
Hello everyone,

currently i'm working on my master thesis and trying to implement a new kinematics class. Before i implemented the changes i tested the normal klipper software and it worked fine with my setup. I think i finished the code development and compiled the micro-controller code for first tests. But when i try to connect to the printer via the OctoPrint Server, the printer can´t connect due to timeout.
I guess a problem appears during compiling "c_helper.so" as you can see in the attached klippy.log. What i changed in the code is the following. All changes that i've made so far are all in the klippy folder and the subfolders.
In the folder "klippy" i made some changes to the Python-Files "gcode.py" and "toolhead.py". In the subfolder "chelper" i added a new Kinematics C-File and changed the Files "trapq.c", "trapq.h" and "__init__.py". And last but not least i added a new Python-File in the subfolder "kinematics".
The printer can connect without adding changed or new developed C-Files. Does that mean, there is only a problem in the "__init__.py" and the C-Files or do I need to make changes somewhere else than in the aforementioned files, too?
I hope you can help me with this problem. Thanks a lot in advance!
[klippy.log](https://github.com/KevinOConnor/klipper/files/4666982/klippy.log)

KevinOConnor, 2020-05-24:
Unfortunately, we don't provide software development advice here.  If the code has been changed and no longer works, you'll need to track down what change caused the regression.

-Kevin

KevinOConnor, 2020-05-25:
FYI, if you are looking for the C compiler errors, then you can follow the steps described at: https://www.klipper3d.org/Debugging.html#translating-gcode-files-to-micro-controller-commands

When running the software from the terminal, the C compiler errors should be shown on that terminal.

-Kevin

Bernouillle, 2020-05-26:
Thanks for your response and your Debugging advice!
I tried to run the commands listed at the Debugging page, but the file "test.gcode" couldn´t be found. Do i have to use an own .gcode file?

Bernouillle, 2020-05-27:
Found a solution for the missing .gcode file! Hopefully the Debugging function can help me find the problem.

closed 2020-06-08 by Bernouillle