Klipper3d/klipper #2909: [Feature Request] Make Klipper able to read internal microSD card of controller board

th33xitus, 2020-05-23:
At the moment it seems like Klipper can't read the internal microSD card of (at least) some controller boards. In my case it is the SKR V1.4 / V1.4 Turbo. 

I know that the initial flash of those devices has to be done with removing the SD card and copying the firmware to the card via PC (like it's written in the docs), then put it back in the controller and restart it. But in case of the SKR V1.4 it is the only possible way to flash....always, even though the MCU was already flashed with Klipper before.

I was told it's due to the "crappy" bootloader BTT put on their boards? I don't know and can't confirm. I just know that Marlin is able to mount them without a problem and that re-flashing the firmware is very easy that way. 

Also i know that it is usually not needed to re-flash Klipper that often (if ever, idk), so the added value of adding that capability to Klipper wouldn't be that high than the value is for Marlin, where you have to recompile and reflash the whole firmware after changing something. The added value in general would probably really only relay on switching firmwares back and forth as you like. The added value when looking at Klipper alone would be, that re-flashing devices like the SKR boards would became a breeze without having to fiddle around with bootloades and such (as i was told how you can make it possible on SKR V1.3 boards). So in short: Would be a comfort feature.

**So my feature request is:**
Is it possible to add that capability of reading the internal microSD card of the controller board to Klipper as well like Marlin does?

KevinOConnor, 2020-05-24:
Possibly, if an interested developer wishes to add this.  There are other bootloaders available that don't have this issue, however.

-Kevin

th33xitus, 2020-05-24:
You are right. I was told i could flash another bootloader, but was also told that if something goes wrong with that you will brick the board. And then you have a fancy paperweight i guess. Having the risk of destroying your hardware would be minimized with that function.

Would be great if there is a developer who wants to add this. Thanks for answering though.

mental405, 2020-05-25:
Are you talking about using it as a data source for gcode to be read?

th33xitus, 2020-05-25:
@mental405 No. Im talking about just being able to mount the internal microSD in the Linux file system of the Pi and therefore having access to the firmware file. Using it for gcode makes no sense at all imo.

dianlight, 2020-05-25:
On SKR / LPC176x (and many other mcu) the SD is connected in SPI. I did some tests and it is possible to create a simple Python driver to read and write the blocks.
To make it available to the host (linux), however, it would be necessary to re-expose it as a device to allow mount / umount.

A lot of work and I don't think the result is stable given the possible reboots of the MCU.

If you are using a USB cable between the RPi and the SKR, replacing the bootloader would be the best solution.

Changing the bootloader is not such a dangerous thing especially on an LPC176x where it can be replaced without a programmer. [The bootloader](https://github.com/Arksine/LPC17xx-DFU-Bootloader/releases/tag/0.4.1)

If instead you use the internal UART on the SKR TFT connector I am also looking for a solution to develop ... maybe adding the autoupate ability in the firmware could be a good idea.

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the conversation has concluded.  It will be up to an interested developer to implement it if they wish.

-Kevin

closed 2020-06-14 by KevinOConnor