Klipper3d/klipper #2911: Problem with SKR1.3 and TMC2209 - Stepper too fare in the past

case-tm77, 2020-05-23:
Hy, i have changed my board from a MKS 1.4 an Ramps 1.4 to a SKR1.3 with TMC2209.

I made every step from the installing guide, checket every stepper, endstops and calibrated the PIDs and the extruder.

Everything seams fine and working like expected.

But when i start a print, soon as the extruder should start extruding, Klipper shoots down with the "Stepper too fare in the past" message.

I made a little search hear and on this search i reduced the microsteps of the extruder form 32 to 16 an after that to 8.

But nothing changed.

Can you pleas help?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4672543/klippy.log)

BlackStump, 2020-05-24:
pressure_advance = 0064    -- is a tad big try disabling pressure advance and try a test print.

follow the steps to config it correctly

https://www.klipper3d.org/Pressure_Advance.html

case-tm77, 2020-05-24:
that explains why my prusa steel with out pressure advance was working fine and the hevo not.

without pressure_advance now it prints nearly perfect.

thank you BlackStump for the hint

closed 2020-05-24 by case-tm77