Klipper3d/klipper #2912: Not possible to control internal pullups for ADC buttons

mk2lehe, 2020-05-24:
I've been happily using MKS GEN L with display from ANET A8 (connected to AUX2, switched cables for 5V and GND). I added a kill-switch function, my config is this:

> [display]
lcd_type: hd44780
rs_pin: ar64
e_pin: ar44
d4_pin: ar63
d5_pin: ar40
d6_pin: ar42
d7_pin: ar65
up_pin: analog12
analog_range_up_pin: 9000, 13000 #0.680
down_pin: analog12
analog_range_down_pin: 800, 1300 #0.174
click_pin: analog12
analog_range_click_pin: 2000, 2500 #0.318
back_pin: analog12
analog_range_back_pin: 4500, 5000 #0.496 
kill_pin: analog12
analog_range_kill_pin: 400, 500  #0.082 value=4700*adc/(1-adc) use left&right as kill pin

Is there a way to control the internal pull-up on ADC buttons, in my case analog12? I do realize I might need to re-tune all analog values on that pin. 
The reason is that if I disconnect the display, the analog12 is floating (mks gen l does not have hardware pull-up on this pin). Last restart in attached klippy.log is with the display cable disconnected. 
I would like to disconnect the display without worrying about floating ADC button - in my case it accidently triggered kill_pin. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4673336/klippy.log)

KevinOConnor, 2020-05-24:
Klipper does not currently support mixing pullups with ADC pins.  It may be possible to do this on some micro-controllers (some micro-controller prohibit this).  If so, and interested developer would need to do the work and propose it.

That said, it's never good to connect or disconnect the display (or any other wires) from a micro-controller while it is powered up.  Doing so can cause lead to permanent damage of the micro-controller and/or display.

-Kevin

mental405, 2020-05-25:
to enable/disable the pullup resistors you need to add a caret to the front of the pin number

ar64 becomes ^ar64

to invert the logic you add a bang to the front of the pin number

ar64 becomes !ar64

you can also combine the two

closed 2020-06-14 by KevinOConnor