Klipper3d/klipper #2913: Connect SKR 1.4 Turbo and RPi3 using RX, TX, GND (UART)

Tomblarom, 2020-05-24:
Hey! Trying to get rid of the clunky USB cable on my printer build [(like in this guide)](https://community.octoprint.org/t/simple-3-wire-serial-connection-from-pi3-to-skr-v1-3-board/10166). Unfortunately I can't establish a connection.. I've been made aware of that there might be interference, since the ribbon cable isn't shielded. Tried to use a USB cable, unfortunately got the same result..

**Klipper `make menuconfig`**:
```
[*] Enable extra low-level configuration options
    Micro-controller Architecture (LPC176x (Smoothieboard))  --->
    Processor model (lpc1769 (120 MHz))  --->
[*] Target board uses Smoothieware bootloader
[ ] Use USB for communication (instead of serial)
(250000) Baud rate for serial port
[ ] Specify a custom step pulse duration
()  GPIO pins to set at micro-controller startup
```

**`/boot/config.txt`:**
```
dtoverlay=pi3-miniuart-bt
enable_uart=1
```

**`/boot/cmdline.txt`:**
```
console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

**`/tmp/klippy.log` (3-pin connector plugged in):**
```
...
Starting serial connect
DWC2:45475.51 - 10.0.0.27 - GET - /rr_connect?password=reprap&time=2020-5-24T20%3A37%3A54
got {'#receive_time': 1239.266556089, u'next_clock': 3207911936L, u'oid': 4, u'value': 32760, '#name': u'analog_in_state', '#sent_time': 1239.245785412}
got {'#receive_time': 1239.3565593179999, u'next_clock': 3218711936L, u'oid': 13, u'value': 32758, '#name': u'analog_in_state', '#sent_time': 1239.3498573910001}
got {'#receive_time': 1239.566568902, u'next_clock': 3243911936L, u'oid': 4, u'value': 32759, '#name': u'analog_in_state', '#sent_time': 1239.556394162}
got {'#receive_time': 1239.6565767139998, u'next_clock': 3254711936L, u'oid': 13, u'value': 32758, '#name': u'analog_in_state', '#sent_time': 1239.608011818}
Loaded MCU 'mcu' 77 commands (v0.8.0-530-g1d2746f1-dirty-20200524_191206-rpi3-3d-hevo / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 BUS_PINS_i2c0=P0.28,P0.27 BUS_PINS_i2c2=P0.11,P0.10 SERIAL_BAUD=250000 STEP_DELAY=2 BUS_PINS_ssp1=P0.8,P0.9,P0.7 BUS_PINS_i2c1a=P0.20,P0.19 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_serial=P0.3,P0.2 RECEIVE_WINDOW=192 MCU=lpc1769 STATS_SUMSQ_BASE=256 CLOCK_FREQ=120000000
MCU error during connect
Traceback (most recent call last):
  File "/home/hevo/klipper/klippy/klippy.py", line 141, in _connect
    cb()
  File "/home/hevo/klipper/klippy/mcu.py", line 553, in _connect
    config_params = self._send_get_config()
  File "/home/hevo/klipper/klippy/mcu.py", line 541, in _send_get_config
    self._name,))
error: Can not update MCU 'mcu' config as it is shutdown
Build file /home/hevo/klipper/klippy/../.config(1104): Sun May 24 19:13:53 2020
========= Last MCU build config =========
...
```

**`/tmp/klippy.log` (3-pin connector pulled out):**
```
...
Starting serial connect
Timeout on serial connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/hevo/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/hevo/klipper/klippy/serialhdl.py", line 186, in send_with_response
    return src.get_response(cmd, self.default_cmd_queue)
  File "/home/hevo/klipper/klippy/serialhdl.py", line 242, in get_response
    self.serial.raw_send_wait_ack(cmd, minclock, minclock, cmd_queue)
  File "/home/hevo/klipper/klippy/serialhdl.py", line 178, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
...
```
[klippy.log](https://github.com/KevinOConnor/klipper/files/4674172/klippy.log)

Tomblarom, 2020-05-25:
Actually I'm not sure what I did, but it's working now.
- removed the klipper folder, recloned the repo
- made the klipper.bin using the settings above, flashed
- rebooted, plugged the serial cable in and it worked!

dianlight, 2020-05-25:
**Cable**:  ->  check on skr.  .
 On Rpi Pin 6,8,10 -> GND (Black), TX (Green), RX (Red)
 On SKR ( I'have a 1.3 so check!) -> +5 (NC), GND (Black), TX(Red),RX(Green), RST(NC) in the image is Green/Red/Black

**Klipper make menuconfig:** -> correct

**/home/pi/printer.cfg:** -> add restart method rst is not connected

```
[mcu]
serial: /dev/ttyAMA0
restart_method: command

```

**/boot/config.txt:** -> enable_uart=1 is optional
`dtoverlay=pi3-miniuart-bt`

**/boot/cmdline.txt:** -> correct

Tomblarom, 2020-05-26:
The wiring of the cables should be correct. 1.4 differs to 1.3:
[image]

Which method is that? How to find it out / define it?

dianlight, 2020-05-26:
Interestingly they turned the connector between versions.  :-S

You say:
> Tried to use a USB cable, unfortunately got the same result..

Does connecting via USB work?

If it doesn't work, sorry for the question but did you make the firmware 'flash' on the mcu?

Try to switch between RX and TX cable. In theory they should be crossed but it is not said that the labels are already inverted to make them correspond.

L.

Tomblarom, 2020-05-26:
I just need to add `restart_method: command`? Thought I had to place `command` with a corresponding command. ^^ Going to test it soon. Found this guy, who did it as well: https://www.thingiverse.com/thing:3851024

I meant, I tried to use a shielded USB cable. Flashing the firmware with `[*] Use USB for communication (instead of serial)` works perfectly fine. Yes it's definitely successfully flashed.

I think I tried this already and the result was, as if the cable is not plugged in.

Oh and btw, I'm not using Ocotoprint! I'm using DWC2 ([Duet Web Control 2 for Klipper](https://klipper.info/klipper-+-dwc2-1/installing-dwc2)) as you see from my klippy.log.

dianlight, 2020-05-26:
`restart_method: command` means that the MCU restart is sent as a command and not by changing the state of the DTR signal.

You don't need a shielded cable, it's a TTL serial signal.

I don't know about DWC2 integration. Try commenting on the [web_dwc2] part and restarting klipper.

One last thing, what bootloader did you use on the SKR?

L.

Tomblarom, 2020-05-26:
Thanks for your guiding comments. Going to test these later.

I havn't changed or upgraded the bootloader on my SKR. I assume the default one (preinstalled)?

Tomblarom, 2020-05-26:
Adding the restart-line done the trick! Had to do a `service klipper restart` and `FIRMWARE_RESTART`, but getting a connection now!

closed 2020-05-27 by Tomblarom