Klipper3d/klipper #2914: klipper didn't work for me

orelias, 2020-05-24:
Hello,so i followed the instructions to the letter and flashed mks gen l board ,now i can't control the printer.I can only go as far as connect via octoprint and i hit a wall.The display is also not working.Please help.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4674364/klippy.zip)

theopensourcerer, 2020-05-24:
Unable to open port: [Errno 2] could not open port
/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0:
[Errno 2] No such file or directory:

orelias, 2020-05-25:
> Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0: [Errno 2] No such file or directory:
> […](#)

Thanks for pointing that,how do i go about resolving that.Please keep in mind i have little to no knowledge about coding.

theopensourcerer, 2020-05-25:
In the FAQ:

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

orelias, 2020-05-28:
> In the FAQ:
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#wheres-my-serial-port

will try it and reply to the issue

orelias, 2020-05-28:
I give up.how can i get passed this issue.tried following the trouble shooting link.didn't work for me
[klippy.log](https://github.com/KevinOConnor/klipper/files/4697704/klippy.log)

theopensourcerer, 2020-05-28:
The error in your log now says:

`"Error: Option 'set_position_z' is not valid in section 'probe'"`

Looking in the example config files, that option belongs in the section called [homing_override]

orelias, 2020-05-29:
so i was able to get klipper "ready" now the printer axis move in the wrong direction(x and y).I did try to remove the exclamation mark in the x and y.and it moves the right direction for some 0.00 something mm and issues an error on the endstop triggers.z axis wont hit home but does moves a few steps in the right direction.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4704194/klippy.log)

orelias, 2020-05-30:
so i was able to fix the issue of axis movement.now i cant print.it keeps killing the extruder temp to zero immediately i commence a print job
[klippy.log](https://github.com/KevinOConnor/klipper/files/4706487/klippy.log)

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like a solution was found for the original question.

-Kevin

closed 2020-06-14 by KevinOConnor

orelias, 2020-06-14:
Hello,
Sadly i was not able to solve it by myself.I went back to marlin.Thanks
anyway

Regards
Mark