Klipper3d/klipper #2916: How to use Rpi’s GPIO?

eziologico, 2020-05-25:
Someone can tell me what is the syntax in printer.cfg file to use the Raspberry Pi GPIO pins and the available options?
I didn't found any documentation.

Thanks

mental405, 2020-05-25:
Are you wanting to use them for serial communication or for like fans and thermistors and such?

eziologico, 2020-05-25:
Hi @mental405 ,
Not for serial communication, for switch on-off light or whatever.
I would like to use the local GPIO from RPi where klipper is running.
I found some posts about it but it's no clear how to do it.

Thanks

dianlight, 2020-05-25:
You need to build/install and start the klipper_mcu and use it in config as secondary mcu.
The startup scripts are in PR #2781 

In prnter.cfg add

```
[mcu host]
serial: /tmp/klipper_host_mcu
```

and then you can use the pins or bus like:

```
[output_pin caselight]
pin: host:P20 

[temperature_sensor enclosure_temp]
sensor_type: HTU21D
i2c_mcu: host
i2c_bus: i2c.1
htu21d_hold_master: False
```

eziologico, 2020-05-26:
Hi @dianlight ,
Thank you for your reply.
I think, in first, I must create the klipper.elf file.
I tried to make a "linux process" firmware but I get an error:

```
  ___  ____  _   _____                ____  _           
 / _ \|  _ \(_) |__  /___ _ __ ___   |  _ \| |_   _ ___ 
| | | | |_) | |   / // _ \ '__/ _ \  | |_) | | | | / __|
| |_| |  __/| |  / /|  __/ | | (_) | |  __/| | |_| \__ \
 \___/|_|   |_| /____\___|_|  \___/  |_|   |_|\__,_|___/
                                                        
Welcome to Armbian buster with Linux 5.4.28-sunxi64

System load:   0.00 0.00 0.00   Up time:       2 days
Memory usage:  21 % of 479MB    Zram usage:    8 % of 239Mb          IP:            192.168.10.32
CPU temp:      50°C             
Usage of /:    19% of 15G       

[ General system configuration (beta): armbian-config ]

Last login: Sun May 24 08:10:21 2020 from 192.168.10.157

pi@opi-kossel:~$ cd klipper
pi@opi-kossel:~/klipper$ make menuconfig
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

pi@opi-kossel:~/klipper$ make
make: ***  Nessuna regola per generare l'obiettivo "out/board/irq.h", necessario per "out/src/sched.o".  Arresto.
pi@opi-kossel:~/klipper$ 
```

I upgraded klipper to the latest git version.

One question: does klipper process still run if the printerboard is offline in order to use a RPi Gpio pin to turn on-off the printer with the M80/M81 gcode?

Thank you for your help,

Ezio

dianlight, 2020-05-26:
It appears that it did not correctly save the .config file. Try to reconfig with menuconfig or nconfig target.

Can you post it?

For M80 / M81 question the current klippy architecture does not allow to command one MCU if the other is in error or in shutdown. So for now you can't have an MCU that controls the other's power switch.

eziologico, 2020-05-26:
Ok, a "make clean" fixed it.
So the linux firmware is running in /usr/local/bin/klipper_mcu with your start script in /etc/init.d and the /tmp/klipper_host_mcu virtual printer is present.

 adding the second mcu in printer.cfg klipper is still running
```
[mcu host]
serial: /tmp/klipper_host_mcu
```
if I define the pin it stops working
```
[output_pin caselight]
pin: host:P20 

Log:
Loaded MCU 'mcu' 80 commands (v0.8.0-535-g5afa5d3f-20200525_150208-opi-kossel / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB3,PB2,PB1 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PE0,PE1 BUS_PINS_twi=PD0,PD1 MCU=atmega2560 STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Starting serial connect
Loaded MCU 'host' 83 commands (v0.8.0-535-g5afa5d3f-20200526_112121-opi-kossel / gcc: (Debian 8.3.0-6) 8.3.0 binutils: (GNU Binutils for Debian) 2.31.1)
MCU 'host' config: PCA9685_MAX=4096 ADC_MAX=4095 PWM_MAX=32768 CLOCK_FREQ=50000000 STEP_DELAY=2 MCU=linux STATS_SUMSQ_BASE=256
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (617 moves)
Sending MCU 'host' printer configuration...
MCU 'host' shutdown: Unable to open out GPIO chip line
Build file /home/pi/klipper/klippy/../.config(712): Tue May 26 11:19:05 2020
========= Last MCU build config =========
#
# Automatically generated file; DO NOT EDIT.
# Klipper Firmware Configuration
#
CONFIG_LOW_LEVEL_OPTIONS=y
# CONFIG_MACH_AVR is not set
# CONFIG_MACH_ATSAM is not set
# CONFIG_MACH_ATSAMD is not set
# CONFIG_MACH_LPC176X is not set
# CONFIG_MACH_STM32 is not set
# CONFIG_MACH_PRU is not set
CONFIG_MACH_LINUX=y
# CONFIG_MACH_SIMU is not set
CONFIG_STEP_DELAY=2
CONFIG_BOARD_DIRECTORY="linux"
CONFIG_CLOCK_FREQ=50000000
CONFIG_LINUX_SELECT=y
# CONFIG_CUSTOM_STEP_DELAY is not set
CONFIG_INITIAL_PINS=""
CONFIG_HAVE_GPIO=y
CONFIG_HAVE_GPIO_ADC=y
CONFIG_HAVE_GPIO_SPI=y
CONFIG_HAVE_GPIO_I2C=y
CONFIG_HAVE_GPIO_HARD_PWM=y
CONFIG_HAVE_GPIO_BITBANGING=y
# CONFIG_HAVE_CHIPID is not set
CONFIG_INLINE_STEPPER_HACK=y
=======================
Build file /home/pi/klipper/klippy/../out/klipper.dict(6646): Tue May 26 11:21:21 2020
Last MCU build version: v0.8.0-535-g5afa5d3f-20200526_112121-opi-kossel
Last MCU build tools: gcc: (Debian 8.3.0-6) 8.3.0 binutils: (GNU Binutils for Debian) 2.31.1
MCU 'mcu' shutdown: Command request

```

I have tried different pins and name without success.

Thank you,

Ezio

dianlight, 2020-05-26:
> MCU 'host' shutdown: Unable to open out GPIO chip line

The GPIO is already used by another process or no gpichip is found.

You are on an OrangePi right?  I don't know the chipset.

### For the gpiochip:
Try `gpiodetect` command and post the result. The klipper implementation use only the **gpiochip0** 

### For the process:
Use the `gpioinfo` command or try  `ls /sys/class/gpio/ `and see if the gpio is exported.

Remember you can't use another process like an Octoprint to command the same GPIO pin.

L.

eziologico, 2020-05-27:
Hi @dianlight ,
To speed up a little bit and to be sure there are not other unwanted process running, I abandoned the OrangePi and did a fresh installation of Raspbian Buster Lite and Klipper on a Raspberry Pi 2 model B.
Using the scripts from your Klipper Fork, I have now the Linux process up and running but Klipper is unable to open the virtual printer /tmp/klipper_host_mcu due to "Permission denied".

```
Starting serial connect

Printer is not ready
The klippy host software is attempting to connect.  Please
retry in a few moments.

Loaded MCU 'mcu' 80 commands (v0.8.0-539-g1ab41cf4-20200527_104711-rpi-kossel / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB3,PB2,PB1 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PE0,PE1 BUS_PINS_twi=PD0,PD1 MCU=atmega2560 STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Starting serial connect
Unable to open port: [Errno 13] Permission denied: '/tmp/klipper_host_mcu'
Unable to open port: [Errno 13] Permission denied: '/tmp/klipper_host_mcu'
Unable to open port: [Errno 13] Permission denied: '/tmp/klipper_host_mcu'
Unable to open port: [Errno 13] Permission denied: '/tmp/klipper_host_mcu'
```

```
ls -la /tmp/

totale 124
drwxrwxrwt  9 root root  4096 mag 27 13:21 .
drwxr-xr-x 21 root root  4096 mag 27 09:20 ..
drwxrwxrwt  2 root root  4096 mag 27 12:17 .font-unix
drwxrwxrwt  2 root root  4096 mag 27 12:17 .ICE-unix
lrwxrwxrwx  1 root root    10 mag 27 12:34 klipper_host_mcu -> /dev/pts/2
-rw-r--r--  1 pi   pi   84325 mag 27 13:23 klippy.log
drwx------  2 pi   pi    4096 mag 27 12:41 mc-pi
lrwxrwxrwx  1 pi   pi      10 mag 27 13:21 printer -> /dev/pts/0
drwx------  3 root root  4096 mag 27 12:17 systemd-private-5e41b4a3ae0f4814b42ad576c51fdae8-systemd-timesyncd.service-Oys2He
drwxrwxrwt  2 root root  4096 mag 27 12:17 .Test-unix
drwxrwxrwt  2 root root  4096 mag 27 12:17 .X11-unix
drwxrwxrwt  2 root root  4096 mag 27 12:17 .XIM-unix
```

The process starts with your klipper-mcu-start.sh script in init.d

Thank you,

Ezio

PS: Your flash_linux.sh script have too many "/", I removed one from "out/"
```
sudo ./scripts/flash-linux.sh
Installing mirco-controller code to /usr/local/bin/
cp: impossibile eseguire stat di 'out//klipper.elf': File o directory non esistente
```

eziologico, 2020-05-27:
Hi @dianlight ,
So the problem is in the /tmp/klipper_host_mcu link that is not owned by pi user but by root and the pi user is not in tty group by default.
As I don't know how to change the link I added the pi user to tty group and it works now.
```
groups pi
pi : pi adm dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio

sudo adduser pi tty
```

It would be nice to have a script like "install-beaglebone.sh" to install everything. Maybe there is already one but is not documented.

For gpioinfo command I installed gpiod package.
I'm using pin 26 as output for test

```
sudo apt install gpiod

sudo gpiodetect 
gpiochip0 [pinctrl-bcm2835] (54 lines)

sudo gpioinfo 
gpiochip0 - 54 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused  output  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed    "klipper"  output  active-high [used]
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed       "led1"   input  active-high [used]
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed       unused   input  active-high 
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused   input  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused   input  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed       "led0"  output  active-high [used]
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
	line  50:      unnamed       unused   input  active-high 
	line  51:      unnamed       unused   input  active-high 
	line  52:      unnamed       unused   input  active-high 
	line  53:      unnamed       unused   input  active-high 

```

Can I use some pins in Linux environment for WebIOPi and others in Klipper?

Thank you for your help and your great work,

Ezio

dianlight, 2020-05-27:
Thanks for reporting the error on `./scripts/flash-linux.sh` (I proceed to the correction as soon as possible). In reality the use case would be the use through make flash.

Being ` klipper_host_mcu -> /dev/pts/2` a link you have to heal the permissions of /dev/pts/2. it Is correct that pi is in the tty group. Usually the addition to the group is done directly by the octoprint installation script so it is not documented ... maybe it should.

A pin can be used by one and only one process. So you can use any unused pins.
**But beware of programs that use libraries such as wiringpi that do not "reserve" the pin on the kernel so see the pin as not used.**

If you are using my branch, manage the second mcu with the parameter MCU=2 so as not to redo the configuration every time.
So you can do:
```
#main mcu
make menuconfig  #only the first time
make
make flash

#mcu secondary
make MCU = 2 menuconfig #only the first time
make MCU = 2
make flash

```

Last thing if you give me the output of `sudo gpiodetect e sudo gpioinfo`
on OrangePi I can understand why it doesn't work and maybe send a correction PR.

L.

eziologico, 2020-05-27:
Hi @dianlight 
This is the output of the OrangePi Zero Plus H5.
I made a lot of test trying to manage the GPIOs from remote but without success, so i thing this system is not more reliable.
I can try again in another OrangePi Zero H2 that is more clean, if you want.
```
  ___  ____  _   _____                ____  _           
 / _ \|  _ \(_) |__  /___ _ __ ___   |  _ \| |_   _ ___ 
| | | | |_) | |   / // _ \ '__/ _ \  | |_) | | | | / __|
| |_| |  __/| |  / /|  __/ | | (_) | |  __/| | |_| \__ \
 \___/|_|   |_| /____\___|_|  \___/  |_|   |_|\__,_|___/
                                                        
Welcome to Armbian buster with Linux 5.4.28-sunxi64

System load:   0.00 0.00 0.00  	Up time:       1 day		
Memory usage:  28 % of 479MB  	Zram usage:    3 % of 239Mb  	IP:            192.168.10.32
CPU temp:      56°C           	
Usage of /:    19% of 15G    	

[ General system configuration (beta): armbian-config ]

Last login: Wed May 27 09:52:38 2020 from 192.168.10.100

pi@opi-kossel:~$ sudo gpiodetect 
gpiochip0 [1f02c00.pinctrl] (32 lines)
gpiochip1 [1c20800.pinctrl] (224 lines)

pi@opi-kossel:~$ sudo gpioinfo 
gpiochip0 - 32 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed   "vdd-cpux"  output  active-high [used]
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed "orangepi:green:pwr" output active-high [used]
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
gpiochip1 - 224 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
[... 195 lines cut ...]
	line 204:      unnamed "usb0_id_det" input active-high [used]
	line 205:      unnamed       unused   input  active-high 
	line 206:      unnamed       unused   input  active-high 
	line 207:      unnamed       unused   input  active-high 
	line 208:      unnamed       unused   input  active-high 
	line 209:      unnamed       unused   input  active-high 
	line 210:      unnamed       unused   input  active-high 
	line 211:      unnamed       unused   input  active-high 
	line 212:      unnamed       unused   input  active-high 
	line 213:      unnamed       unused   input  active-high 
	line 214:      unnamed       unused   input  active-high 
	line 215:      unnamed       unused   input  active-high 
	line 216:      unnamed       unused   input  active-high 
	line 217:      unnamed       unused   input  active-high 
	line 218:      unnamed       unused   input  active-high 
	line 219:      unnamed       unused   input  active-high 
	line 220:      unnamed       unused   input  active-high 
	line 221:      unnamed       unused   input  active-high 
	line 222:      unnamed       unused   input  active-high 
	line 223:      unnamed       unused   input  active-high
```

Thanks,

Ezio

eziologico, 2020-05-28:
Hi @dianlight ,
I installed it on the OrangePi Zero H2+ too, and it works!

The pinout is different from RPi
```
pi@opi-kossel:~$ gpio readall
 +------+-----+----------+------+---+ ZEROPLUS +---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
 |   12 |   0 |    SDA.0 |  OFF | 0 |  3 || 4  |   |      | 5V       |     |      |
 |   11 |   1 |    SCL.0 |  OFF | 0 |  5 || 6  |   |      | GND      |     |      |
 |    6 |   2 |      PA6 |  OFF | 0 |  7 || 8  | 0 | OFF  | TXD.1    | 3   | 198  |
 |      |     |      GND |      |   |  9 || 10 | 0 | OFF  | RXD.1    | 4   | 199  |
 |    1 |   5 |    RXD.2 |  OFF | 0 | 11 || 12 | 0 | OFF  | PA07     | 6   | 7    |
 |    0 |   7 |    TXD.2 |  OFF | 0 | 13 || 14 |   |      | GND      |     |      |
 |    3 |   8 |    CTS.2 |  OFF | 0 | 15 || 16 | 0 | OFF  | SDA.1    | 9   | 19   |
 |      |     |     3.3V |      |   | 17 || 18 | 0 | OFF  | SCL.1    | 10  | 18   |
 |   15 |  11 |   MOSI.1 |  OFF | 0 | 19 || 20 |   |      | GND      |     |      |
 |   16 |  12 |   MISO.1 |  OFF | 0 | 21 || 22 | 0 | OFF  | RTS.2    | 13  | 2    |
 |   14 |  14 |   SCLK.1 |  OFF | 0 | 23 || 24 | 0 | OFF  | CE.1     | 15  | 13   |
 |      |     |      GND |      |   | 25 || 26 | 0 | OFF  | PA10     | 16  | 10   |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+ ZEROPLUS +---+------+----------+-----+------+
```

This is the output of the OrangePi Zero H2+.
```
  ___  ____  _   _____              
 / _ \|  _ \(_) |__  /___ _ __ ___  
| | | | |_) | |   / // _ \ '__/ _ \ 
| |_| |  __/| |  / /|  __/ | | (_) |
 \___/|_|   |_| /____\___|_|  \___/ 
                                    
Welcome to Armbian buster with Linux 5.4.35-sunxi

System load:   0.68 0.21 0.07  	Up time:       0 min		
Memory usage:  18 % of 491MB  	IP:            192.168.10.31
CPU temp:      49°C           	
Usage of /:    16% of 15G    	

Last login: Thu May 28 07:59:09 2020 from 192.168.10.100

pi@opi-ender3:~$ sudo gpiodetect 
gpiochip0 [1c20800.pinctrl] (224 lines)
gpiochip1 [1f02c00.pinctrl] (32 lines)
gpiochip2 [ftdi-cbus] (4 lines)

pi@opi-ender3:~$ sudo gpioinfo 
gpiochip0 - 224 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed    "klipper"  output  active-high [used]
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed "orangepi:red:status" output active-high [used]
	line  18:      unnamed       unused   input  active-high 
[... 223 lines cut ...]
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
gpiochip2 - 4 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
```

Thanks,

Ezio

dianlight, 2020-05-28:
The problem on " OrangePi Zero Plus H5" is the sequence of gpiochip. Actual implementation use only the gpiochip0 and for the "Opi ZP H5" the right is gpiochip1.

If you are interested and you can test for me I can create a PR to use the other chips too.

eziologico, 2020-05-28:
Yes, I can install armbian from scratch on the OrangePi Zero Plus H5 and test it for you.

If you prefer you can send me a PM,

Ezio

dianlight, 2020-05-29:
I release a very preliminary version of the patch at [dianlight/klipper](https://github.com/dianlight/klipper/tree/linux_gpio_morechip).

**NOTE:** The pin numbers now are declared as P<chipId>_<pinId> so your old P20 become P0_20.

For "Opi ZP H5" where the gpiochip1 need to be use, you can now  assign P1_<gpio> as you can see on gpioinfo output.

I'm waiting for your feedback to clean up the code to propose an official PR.

L.

dianlight, 2020-05-30:
Hi @eziologico ,
  Sorry, after writing the instructions I made a refactor to bring the names of the pins back to the standard.

I did another push on the branch where I should have corrected the problem (if you can do another test I would be grateful) and I optimized the code a bit. 

Also do a make oldconfig I changed the `make scritpts` a bit.

L.

eziologico, 2020-05-30:
hi @dianlight ,
I fetched the new version and declared 2 gpiochips in the linux-firmware as the available gpio pins are on chip1 on this board, anyway the gpiodetect and gpioinfo commands are not installed by default, so it's not so easy to know which chip is the right one.
I don't have any /dev/gpiochip* device in armbian like in raspios, maybe somthing in your script doesn't work as expected.
I tested all the available pins an they works (P1.0 to P1.20 and P1.198 - P1.199), even with a Led on random outputs switching it on/off.
Even declating some P0.0... pins on chip0 seems working but as they aren't available I don't care it.

Orange Pi Zero Plus (H5)
```
 +------+-----+----------+------+---+ ZEROPLUS +---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 |      |     |     3.3V |      |   |  1 || 2  |   |      | 5V       |     |      |
 |   12 |   0 |    SDA.0 | ALT2 | 0 |  3 || 4  |   |      | 5V       |     |      |
 |   11 |   1 |    SCL.0 | ALT2 | 0 |  5 || 6  |   |      | GND      |     |      |
 |    6 |   2 |      PA6 |  OFF | 0 |  7 || 8  | 0 | ALT2 | TXD.1    | 3   | 198  |
 |      |     |      GND |      |   |  9 || 10 | 0 | ALT2 | RXD.1    | 4   | 199  |
 |    1 |   5 |    RXD.2 | ALT2 | 0 | 11 || 12 | 0 | OFF  | PA07     | 6   | 7    |
 |    0 |   7 |    TXD.2 | ALT2 | 0 | 13 || 14 |   |      | GND      |     |      |
 |    3 |   8 |    CTS.2 |  OFF | 0 | 15 || 16 | 0 | ALT3 | SDA.1    | 9   | 19   |
 |      |     |     3.3V |      |   | 17 || 18 | 0 | ALT3 | SCL.1    | 10  | 18   |
 |   15 |  11 |   MOSI.1 | ALT2 | 0 | 19 || 20 |   |      | GND      |     |      |
 |   16 |  12 |   MISO.1 | ALT2 | 0 | 21 || 22 | 0 | OFF  | RTS.2    | 13  | 2    |
 |   14 |  14 |   SCLK.1 | ALT2 | 0 | 23 || 24 | 0 | ALT2 | CE.1     | 15  | 13   |
 |      |     |      GND |      |   | 25 || 26 | 0 | OFF  | PA10     | 16  | 10   |
 +------+-----+----------+------+---+----++----+---+------+----------+-----+------+
 | GPIO | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | GPIO |
 +------+-----+----------+------+---+ ZEROPLUS +---+------+----------+-----+------+``

```

pi@test-gpio:~$ sudo gpioinfo gpiochip1
gpiochip1 - 224 lines:
	line   0:      unnamed       unused  output  active-high 
	line   1:      unnamed       unused  output  active-high 
	line   2:      unnamed       unused  output  active-high 
	line   3:      unnamed       unused  output  active-high 
	line   4:      unnamed       unused  output  active-high 
	line   5:      unnamed       unused  output  active-high 
	line   6:      unnamed       unused  output  active-high 
	line   7:      unnamed       unused  output  active-high 
	line   8:      unnamed       unused  output  active-high 
	line   9:      unnamed       unused  output  active-high 
	line  10:      unnamed       unused  output  active-high 
	line  11:      unnamed       unused  output  active-high 
	line  12:      unnamed       unused  output  active-high 
	line  13:      unnamed       unused  output  active-high 
	line  14:      unnamed       unused  output  active-high 
	line  15:      unnamed       unused  output  active-high 
	line  16:      unnamed       unused  output  active-high 
	line  17:      unnamed "orangepi:red:status" output active-high [used]
	line  18:      unnamed       unused  output  active-high 
	line  19:      unnamed       unused  output  active-high 
	line  20:      unnamed       unused  output  active-high 
	line  21:      unnamed       unused   input  active-high 
	...
	line 101:      unnamed       unused   input  active-high 
	line 102:      unnamed   "gmac-3v3"  output  active-high [used]
	line 103:      unnamed       unused   input  active-high 
	...
	line 165:      unnamed       unused   input  active-high 
	line 166:      unnamed         "cd"   input   active-low [used]
	line 167:      unnamed       unused   input  active-high 
	...
	line 197:      unnamed       unused   input  active-high 
	line 198:      unnamed       unused  output  active-high 
	line 199:      unnamed    "klipper"  output  active-high [used]
	line 200:      unnamed       unused   input  active-high 
	...
	line 203:      unnamed       unused   input  active-high 
	line 204:      unnamed "usb0_id_det" input active-high [used]
	line 205:      unnamed       unused   input  active-high 
	...
	line 223:      unnamed       unused   input  active-high 
```

At the moment the configuration is not so user frendly, can it be improved?

Ezio

dianlight, 2020-05-30:
hi @eziologico ,
 summarizing is now going but configuration is a nightmare? ;)

> ...anyway the gpiodetect and gpioinfo commands are not installed by default, so it's not so   easy to know which chip is the right one.
> ...

Did you use the install-debian.sh script? 

The kernel interface allows control of the gpio chips as these are electrically connected and which pins are exposed depends on the vendor and the pcb design (in some cases also on the pcb version).
Many libraries have pre-designed configurations for the most common boards but it would not be logical for a project like klipper.

What you can do is add the config examples for the various boards with the pins exposed to the documentation exactly as it is done for the mcu. ( Kevin allowing volunteers to be accepted;) )

> ...
> I don't have any /dev/gpiochip* device in armbian like in raspios, maybe somthing in your script doesn't work as expected.
> ...
> 

`/dev/gpiochip*` is the character device that is also opened by the mcu therefore it is very strange that it does not exist. The system configuration probably loads the kernel module of the gpio chips only when someone tries to access it. Try doing an `ls /dev/gpiochip*` after the MCU has started and you've tested at least one pin or after a `gpioinfo`

If the devices are there you can try after a reboot when they do not appear in /dev to run the command `sudo stat /dev/gpiochip0` and see if they appear or if only the 0 appears.

L.

eziologico, 2020-05-30:
Hi @dianlight ,
So it seems working now. :)

> Did you use the install-debian.sh script?

No, I always use ```install-octopi.sh``` as in the Klipper documentetion and ```flash-linux.sh``` to install the linux firmware.
I don't know what the other scripts are for.
Should I use ```install-debian.sh``` script?

>Try doing an ```ls /dev/gpiochip*``` after the MCU has started and you've tested at least one pin or after a gpioinfo

I checked it again and they appear now, I don't know why it wasn't before.

```
pi@test-gpio:~$ ls -la /dev/gpiochip*
crw------- 1 root root 254, 0 May 30 11:13 /dev/gpiochip0
crw------- 1 root root 254, 1 May 30 11:13 /dev/gpiochip1
crw------- 1 root root 254, 2 May 30 11:13 /dev/gpiochip2
```

Ezio

dianlight, 2020-05-30:
PR #2935

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-06-14 by KevinOConnor