Klipper3d/klipper #2920: MCU shutdown: Watchdog timer!

supermerill, 2020-05-26:
Hello
My printer stopped printing almost at the end of a big print.
From what i can read it's due to the watchdog timer. I tried to find something in the documentation about it, to know what i need to do to avoid this problem is the future, but i can't find anything.
Note that after a firmware_restart, it's working again, like nothing happen.
What does it means? The mcu stopped working, so i should change it?
does this have something to do with [that](https://github.com/KevinOConnor/klipper/blob/1eb2d4da902867b3bf4fdee1b2d272c170f93468/docs/FAQ.md#when-i-set-restart_methodcommand-my-avr-device-just-hangs-on-a-restart)?

setup: voron2: 2*arduino+ramps+a4988, rpi3+
installed on ced389f4f635ed362eccbddba19e4e54dd96e7d6 commit

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4681290/klippy.zip)

Thank you for your great work!

KevinOConnor, 2020-06-01:
The "watchdog timer" is a hardware function that verifies the software isn't in a "locked up state".  It looks like some internal error happened on the "z mcu", which the watchdog hardware detected and alerted on.  I've never seen anyone report this error before.  Things I'd check for are possible voltage issues on the z mcu and/or possible overheating of that mcu.

-Kevin

supermerill, 2020-06-01:
That was after ~20h of print so maybe a bit of overheating, i'll add a fan, thank you!

closed 2020-06-01 by supermerill